pschatzmann/arduino-audio-tools #1848: RP2040 Player

pierrotm777, 2024-12-29:
### Problem Description

Try to use the example https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sdfat-analog with my RP2040

### Device Description

Waveshare RP2040 Zero

### Sketch

```cpp
https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sdfat-analog
```

### What is your development environment

W11

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pierrotm777, 2024-12-29:
My compilation error:
`In file included from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SDFS\src/SDFS.h:35,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SD\src/SD.h:24,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/AudioSourceSD.h:7,
                 from C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-15920-a4gs48.0eg2\player-sdfat-analog\player-sdfat-analog.ino:11:
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/SdFat.h:462:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
  462 | #warning File not defined because __has_include(FS.h)
      |  ^~~~~~~
In file included from C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/ArduinoAPI.h:31,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\cores\rp2040/api/ArduinoAPI.h:2,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\cores\rp2040/Arduino.h:28,
                 from C:\Users\pierrot\AppData\Local\arduino\sketches\1654A63FBF84FF2B2275A97D42D18813\sketch\player-sdfat-analog.ino.cpp:1:
c:/tmp/Arduino_Compilateur/arduino-1.8.19/MyArduinoSketch/libraries/SdFat/src/common/ArduinoFiles.h: In instantiation of 'class PrintFile<FsBaseFile>':
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/SdFat.h:468:23:   required from here
  468 | class SdFile : public PrintFile<SdBaseFile> {
      |                       ^~~~~~~~~~~~~~~~~~~~~
C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/Print.h:54:20: warning: 'virtual size_t arduino::Print::write(const uint8_t*, size_t)' was hidden [-Woverloaded-virtual=]
   54 |     virtual size_t write(const uint8_t *buffer, size_t size);
      |                    ^~~~~
In file included from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/ExFatLib/ExFatFile.h:883,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/ExFatLib/ExFatVolume.h:27,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/ExFatLib/ExFatLib.h:28,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/SdFat.h:31:
c:/tmp/Arduino_Compilateur/arduino-1.8.19/MyArduinoSketch/libraries/SdFat/src/common/ArduinoFiles.h:53:10: note:   by 'PrintFile<FsBaseFile>::write'
   53 |   size_t write(uint8_t b) { return BaseFile::write(&b, 1); }
      |          ^~~~~
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SDFS\src/SDFS.h: In member function 'virtual bool sdfs::SDFSImpl::stat(const char*, fs::FSStat*)':
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SDFS\src/SDFS.h:157:36: error: no match for 'operator=' (operand types are 'File32' and 'FsFile')
  157 |         f = _fs.open(path, O_RDONLY);
      |                                    ^
In file included from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/FatLib/FatVolume.h:27,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/FatLib/FatLib.h:28,
                 from c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/SdFat.h:32:
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/FatLib/FatFile.h:1088:7: note: candidate: 'File32& File32::operator=(File32&&)'
 1088 | class File32 : public StreamFile<FatFile, uint32_t> {
      |       ^~~~~~
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat\src/FatLib/FatFile.h:1088:7: note:   no known conversion for argument 1 from 'FsFile' to 'File32&&'
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SDFS\src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SDFS\src/SDFS.h:287:31: error: 'using std::__shared_ptr_access<File32, __gnu_cxx::_S_single, false, false>::element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  287 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-15920-a4gs48.0eg2\player-sdfat-analog\player-sdfat-analog.ino: At global scope:
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-15920-a4gs48.0eg2\player-sdfat-analog\player-sdfat-analog.ino:18:1: error: 'AnalogAudioStream' does not name a type
   18 | AnalogAudioStream out;
      | ^~~~~~~~~~~~~~~~~
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-15920-a4gs48.0eg2\player-sdfat-analog\player-sdfat-analog.ino:20:28: error: 'out' was not declared in this scope
   20 | AudioPlayer player(source, out, decoder);
      |                            ^~~
[... cut ...]
Utilisation de la bibliothèque SPI version 1.0 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SPI 
Utilisation de la bibliothèque SdFat version 2.2.3 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\SdFat 
Utilisation de la bibliothèque libhelix version 0.1 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\arduino-libhelix 
exit status 1

Compilation error: 'AnalogAudioStream' does not name a type`

pschatzmann, 2024-12-29:
I think you are completly wrong here! 
If the SDFAT library is not working on a RP2040 you need to complain with the RP2040 core or the SDFAT library!
Why don't you just use the SD library which comes with the RP2040 ?

closed 2024-12-29 by pschatzmann

pierrotm777, 2024-12-29:
Sorry, i have forgotten to remove my SDFat lib:
I have replaced SDFat by SD

`#include "AudioTools.h"
#include "AudioLibs/AudioSourceSD.h"
#include "AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSD source(startFilePath, ext);
AnalogAudioStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);`

my new errors:
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-18060-1m54rxo.5ubp\player-sdfat-analog\player-sdfat-analog.ino:18:1: error: 'AnalogAudioStream' does not name a type
   18 | AnalogAudioStream out;
      | ^~~~~~~~~~~~~~~~~
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-18060-1m54rxo.5ubp\player-sdfat-analog\player-sdfat-analog.ino:20:28: error: 'out' was not declared in this scope
   20 | AudioPlayer player(source, out, decoder);
      |                            ^~~
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-18060-1m54rxo.5ubp\player-sdfat-analog\player-sdfat-analog.ino: In function 'void setup()':
C:\Users\pierrot\AppData\Local\Temp\.arduinoIDE-unsaved20241129-18060-1m54rxo.5ubp\player-sdfat-analog\player-sdfat-analog.ino:35:14: error: 'out' was not declared in this scope
   35 |   auto cfg = out.defaultConfig();
      |              ^~~
In file included from C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/String.h:33,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\cores\rp2040/api/String.h:2,
                 from C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/IPAddress.h:23,
                 from C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/ArduinoAPI.h:30,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\cores\rp2040/api/ArduinoAPI.h:2,
                 from C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\cores\rp2040/Arduino.h:28,
                 from C:\Users\pierrot\AppData\Local\arduino\sketches\360A7CC3692D035F7CF1E4E83DF3F9D0\sketch\player-sdfat-analog.ino.cpp:1:
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/SDDirect.h: In instantiation of 'void audio_tools::SDDirect<SDT, FileT>::listDir(const char*) [with SDT = SDClass; FileT = fs::File]':
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/SDDirect.h:40:5:   required from 'const char* audio_tools::SDDirect<SDT, FileT>::operator[](int) [with SDT = SDClass; FileT = fs::File]'
   40 |     listDir(start_dir);
      |     ^~~~~~~
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/AudioSourceSD.h:70:26:   required from here
   70 |     file_name = idx[index];
      |                          ^
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/SDDirect.h:107:16: warning: format '%d' expects argument of type 'int', but argument 5 has type 'int32_t' {aka 'long int'} [-Wformat=]
  107 |           LOGD("File %s at index: %d", fn, actual_idx);
      |                                            ~
      |                                            |
      |                                            int32_t {aka long int}
C:/Users/pierrot/AppData/Local/Arduino15/packages/rp2040/hardware/rp2040/4.4.0/ArduinoCore-API/api/deprecated-avr-comp/avr/pgmspace.h:34:20: note: in definition of macro 'PSTR'
   34 | #define PSTR(str) (str)
      |                    ^~~
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioTools/AudioLogger.h:213:83: note: in expansion of macro 'LOG_OUT_PGMEM'
  213 | #define LOGD(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Debug) { LOG_OUT_PGMEM(AudioLogger::Debug, fmt, ##__VA_ARGS__);}
      |                                                                                   ^~~~~~~~~~~~~
c:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040\src/AudioLibs/SDDirect.h:107:11: note: in expansion of macro 'LOGD'
  107 |           LOGD("File %s at index: %d", fn, actual_idx);
      |           ^~~~
Plusieurs bibliothèque trouvées pour "AudioTools.h"
  Utilisé: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040
  Non utilisé: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\arduino-audio-tools
Utilisation de la bibliothèque audio-tools version 0.9.8 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\Arduino_AudioTools-ESP32_RP2040 
Utilisation de la bibliothèque FreeRTOS version 1.0.0 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\FreeRTOS 
Utilisation de la bibliothèque I2S version 2.0 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\I2S 
Utilisation de la bibliothèque AudioBufferManager version 1.0.0 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\AudioBufferManager 
[... cut ...]
Utilisation de la bibliothèque SPI version 1.0 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\SPI 
Utilisation de la bibliothèque ESP8266SdFat version 2.2.2 dans le dossier: C:\Users\pierrot\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.4.0\libraries\ESP8266SdFat 
Utilisation de la bibliothèque libhelix version 0.1 dans le dossier: C:\tmp\Arduino_Compilateur\arduino-1.8.19\MyArduinoSketch\libraries\arduino-libhelix 
exit status 1

Compilation error: 'AnalogAudioStream' does not name a type

pschatzmann, 2024-12-29:
The RP2040 has no internal DAC! 
Use an I2S DAC or PWM...

pierrotm777, 2024-12-29:
I should like PWM for have an analogic output signal.
It's possible with your lib ?

pschatzmann, 2024-12-29:
Please read the corresponding [Wiki](https://github.com/pschatzmann/arduino-audio-tools/wiki/Audio-Input-and-Output)

pierrotm777, 2024-12-29:
Ok, thanks, but my need would be to read the mp3 file from a SD card and use your PWM example after.
Is it possible ?

pschatzmann, 2024-12-29:
Theoretically I don't see any reason why this should not work: just replace the outptu class

pierrotm777, 2024-12-29:
[AudioTools_RP2040_Test.ino.txt](https://github.com/user-attachments/files/18269289/AudioTools_RP2040_Test.ino.txt)
This is my test sketch, but i have errors :-(