pschatzmann/arduino-audio-tools #1867: AudioPlayer not working with Flac?

madskjeldgaard, 2025-01-14:
### Problem Description

I am running a simple, modified AudioPlayer example. It runs fine with Wav but it won't compile with Flac, saying the constructor is wrong:

```
src/main.cpp:62:50: error: no matching function for call to 'audio_tools::AudioPlayer::AudioPlayer(audio_tools::AudioSourceSDFAT&, audio_tools::I2SStream&, audio_tools::FLACDecoder&)'
```

with a WAVDecoder it's fine, but it seems that FLACDecoder and WavDecoder are not the same base type. Maybe this is per design? 

### Device Description

Raspberry Pi Pico 2

### Sketch

```cpp
/*

This sketch simple loops through all flac files in the "/audiofiles" folder on
the SD card and plays them one after the other.

*/
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"

#include "AudioTools/AudioCodecs/CodecFLAC.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include <Adafruit_NeoPixel.h>

constexpr auto logLevel = AudioLogger::Error;

constexpr auto LED_PIN = 22;
constexpr auto NUM_LEDS = 1;
constexpr auto LED_BRIGHTNESS = 100;
Adafruit_NeoPixel strip(NUM_LEDS, LED_PIN, NEO_GRB + NEO_KHZ800);

// SPI pins
constexpr auto csPin = 13;
constexpr auto misoPin = 12;
constexpr auto mosiPin = 11;
constexpr auto sckPin = 10;

// Performance settings
constexpr auto i2s_buffer_size = 4096;
constexpr auto i2s_buffer_count = 6;

// Pico 1 is not powerful enough to go above 44.1khz with this sketch
#if PICONUM == 1
constexpr auto sample_rate = 44100;
#elif PICONUM == 2
constexpr auto sample_rate = 96000;
#else
#pragma message "PICONUM not defined, using default sample rate of 44100")
constexpr auto sample_rate = 44100;
#endif

constexpr auto channels = 2;
constexpr auto bits_per_sample = 16;
constexpr auto baudrate = 115200;

// SDFat
const char *audioFileFolder = "/audiofiles";
const char *ext = "flac";

// Initialize SPI1
SdSpiConfig spi1Config(csPin, DEDICATED_SPI, SPI_CLOCK, &SPI1);
AudioSourceSDFAT source(audioFileFolder, ext, spi1Config);

// Audio setup
AudioInfo info(sample_rate, channels, bits_per_sample);
I2SStream i2sOutput;

// WAVDecoder wavDecoder;
FLACDecoder flacDecoder;

AudioPlayer player(source, i2sOutput, flacDecoder);

// Use the Pico's BOOTSEL button as a trigger to play the next file
auto bootselButtonState = false;
void updateBootSelButton() {
  static bool lastButtonState = HIGH; // Assume button is not pressed initially
  bool currentButtonState = BOOTSEL;

  if (lastButtonState == HIGH && currentButtonState == LOW) {
    LOGI("Bootsel button pressed");
    const auto active = player.next();

    if (!active) {
      LOGI("No more files to play");
      player.begin();
    }
  }

  lastButtonState = currentButtonState;
}

void callbackPrintMetaData(MetaDataType type, const char *str, int len) {
  LOGI("==> ");
  LOGI(toStr(type));
  LOGI(": ");
  LOGI(str);
}

void setup() {
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  SPI1.setCS(csPin);
  SPI1.setMISO(misoPin);
  SPI1.setMOSI(mosiPin);
  SPI1.setSCK(sckPin);

  AudioLogger::instance().begin(Serial, logLevel);

  // setup I2S output
  auto config = i2sOutput.defaultConfig(TX_MODE);

  config.copyFrom(info);
  config.buffer_size = i2s_buffer_size;
  config.buffer_count = i2s_buffer_count;

  auto result = i2sOutput.begin(config);

  if (!result) {
    LOGE("Failed to start I2S output");

    // Set LED to red
    strip.fill(strip.Color(255, 0, 0), 0, NUM_LEDS);
    strip.show();
    while (1)
      ;
  } else {
    LOGI("Successfully started I2S");

    // Set LED to green
    strip.fill(strip.Color(0, 255, 0), 0, NUM_LEDS);
    strip.show();
  }
  source.begin();
  flacDecoder.begin();
  // mp3Decoder.begin();

  LOGI("Starting audio player");

  // Start audio player
  // source.setFileFilter("EX001*");
  player.setBufferSize(i2s_buffer_size);
  player.setAudioSource(source);
  player.setMetadataCallback(callbackPrintMetaData);
  player.setAutoNext(true);
  player.setAutoFade(true);
  player.setVolume(0.5f);
  player.setPath(audioFileFolder);
  player.begin();

  // fill buffer with some data
  player.getStreamCopy().copyN(5);

  const auto numFiles = source.size();
  LOGI("Number of files: %d", numFiles);
  if (numFiles == 0) {
    LOGE("No audio files found");

    // Set LED to blink orange
    while (1) {
      strip.fill(strip.Color(255, 165, 0), 0, NUM_LEDS);
      strip.show();
      delay(100);
      strip.fill(strip.Color(0, 0, 0), 0, NUM_LEDS);
      strip.show();
      delay(100);
    }
  }

  bootselButtonState = BOOTSEL;
  LOGI("Player started :))))))))))))");

  strip.fill(strip.Color(0, 255, 0), 0, NUM_LEDS);
  strip.setBrightness(LED_BRIGHTNESS);
  strip.show();
}

void loop() {
  player.copy();

  // Check if it's inactive and has reached the last file, if so, reset
  if (!player.isActive()) {
    if (!player.next()) {
      LOGI("No more files to play, starting over");
      player.begin();
    }
  }

  updateBootSelButton();
}
```

### What is your development environment

Platformio on MacOS. 

```ini
[platformio]
description = Firmware for a low cost .FLAC-file player
default_envs = raspberrypi-pico

[env]
framework = arduino

build_unflags = -std=gnu++11 -Os
build_flags = -std=gnu++17 -O3 -Wno-unused-variable -Wno-unused-but-set-variable -Wno-unused-function -Wno-format-extra-args -Wsign-compare -Wuninitialized -Wunused-parameter -Wunused

; Use clang-tidy when runnning `pio check`
; https://docs.platformio.org/en/stable/advanced/static-code-analysis/tools/clang-tidy.html
check_tool = clangtidy

; Serial monitor speed, make sure this is matched in Serial.begin()
monitor_speed = 115200

lib_deps =
  SD

  adafruit/Adafruit NeoPixel@^1.12.2

  # SdFat

  # SPI
  SPI

  # I2C
  # Wire

  ; Audio Tools
  ; This library has a TON of audio related stuff
  https://github.com/pschatzmann/arduino-audio-tools.git

  ; Helix MP3 decoder for playing MP3 files
  https://github.com/pschatzmann/arduino-libhelix

  ; Flac support
  https://github.com/pschatzmann/arduino-libflac
  https://github.com/pschatzmann/arduino-libopus

; Pico 1
[env:raspberrypi-pico]
; The Raspberry Pi Pico
; Link: https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html
; For more information about the platformio support:
; https://arduino-pico.readthedocs.io/en/latest/platformio.html
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
board = pico
board_build.mcu = rp2040
# board_build.f_cpu = 133000000L
# Overclocked to 270MHz
board_build.f_cpu = 270000000L
# 200 mhz
# board_build.f_cpu = 200000000L
upload_protocol = picotool

build_flags =
  ${env.build_flags}
  -DPICONUM=1
lib_deps =
  ${env.lib_deps}

; Pico 2
[env:rpipico2]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
# platform = raspberrypi
board = rpipico2
board_build.core = earlephilhower
board_build.f_cpu = 300000000L
build_flags =
  ${env.build_flags}
  -DPICONUM=2
lib_deps =
  ${env.lib_deps}
```

### I have checked existing issues, discussions and online documentation

- [X] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-14:
Just read the Wiki about Codecs: it contains all necessary information!

closed 2025-01-14 by pschatzmann

madskjeldgaard, 2025-01-14:
Okay thanks 🙏