pschatzmann/arduino-audio-tools #1869: Probleme in StreamCopy

jpchamp, 2025-01-18:
### Problem Description

Hello,
I try to connect a INMP441 to a Bluetooth headphone with an ESP32. But the ConverterFillLeftAndRight seen to not work.
I test with the CsvOutput and a channel is alway at 0 (the script).
I use the debug log and i see CoreAudio/BaseConverter.h and CoreAudio/StreamCopy.h.
In StreamCopy and it's call to ConverterFillLeftAndRight<int32_t> filler in "copy", i see in inline size_t copyBytes(size_t bytes)

            size_t result = 0;
..
.. no other modifications of result
..
            if (p_converter!=nullptr) p_converter->convert((uint8_t*)buffer.data(),  result );

I think that converter was alway call with a buffer length = 0.

I try

            if (p_converter!=nullptr) p_converter->convert((uint8_t*)buffer.data(),  bytes_read );

with bytes_read is the result of from->readBytes() and it seen to be OK but i am not sure that the best correction.

Could you examine this problem ?

Best regard.

### Device Description

ESP32 and INMP441

### Sketch

```cpp
#include "AudioTools.h"

AudioInfo to(44100, 2, 16);
   
I2SStream in_i2sStream; // Access I2S as stream
CsvOutput<int16_t> out_csvStream(Serial);
ConverterFillLeftAndRight<int32_t> filler(RightIsEmpty);
NumberFormatConverterStream conv(out_csvStream);
StreamCopy copier(conv, in_i2sStream); // copy i2sStream to csvStream

// Arduino Setup
void setup(void) {
    Serial.begin(115200);
//    AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Debug);
    AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);
    
    auto in_cfg = in_i2sStream.defaultConfig(RX_MODE);
    //cfg.copyFrom(info);
    in_cfg.i2s_format = I2S_STD_FORMAT; // optional because default setting
    in_cfg.bits_per_sample = 32;
    in_cfg.channels = 2; // optional because default setting
    in_cfg.sample_rate = 44100;
    in_cfg.is_master = true; // optional because default setting
    in_cfg.use_apll = false;  // try with yes
    in_cfg.pin_mck = 3; 
    in_i2sStream.begin(in_cfg);

    conv.begin(in_cfg, to);
    out_csvStream.begin(to);
}

// Arduino loop - copy data
void loop() {
    copier.copy(filler);
}
```

### What is your development environment

Arduino

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-18:
I don't think that there is any issue: this has always been working.
Show the output of the Serial Plotter of a recording of a sine wave w/o and with applying the ConverterFillLeftAndRight

jpchamp, 2025-01-19:
Hello
Here is the hardcopy off the plotter and a sinewave with that you said in our Wiki, with/without ConverterFillLeftAndRight and with the StreamCopy.h of GitHub and with the version with the modification of p_converter (this is the only with the 2 channel red/blue superimposed). 
[image]
[image]
[image]
[image]
I have ask I2S for 2 channels because the datasheet of the INMP441 said that there is 2 channels but on at 0, and on some discussions the configuration of this micro is with 2 channels. The ConverterFillLeftAndRight seemed to be the rigth thing to have 2 channles.
Best regards jpc

pschatzmann, 2025-01-19:
I confirm: it should be 

                if (p_converter!=nullptr) p_converter->convert((uint8_t*)buffer.data(),  bytes_read );

Not sure how this bug has been introduced...

ps. If you use ConverterFillLeftAndRight, you can set i2s to 16 bits and save yourself the conversion step to 16 bits

pschatzmann, 2025-01-19:
Maybe there is an issue with 32 bits with ConverterFillLeftAndRight? Try to change I2s and the ConverterFillLeftAndRight to 16 bits.
This has also the advantage that you don't need a coverter step.
This will work for sure !

I committed the correction...

closed 2025-01-19 by pschatzmann