pschatzmann/arduino-audio-tools #1883: Rpi Pico SD card stuttering

trybula, 2025-01-25:
### Problem Description

I have absolutely no problems with streaming music through http, however when I'm trying to play (the exact same file as streamed) from sd card it stutters a lot and is impossible to enjoy.
 I tried different buffer settings, overclocking, SPI frequencies, other sd cards and sd libraries (SD and SDfat), but nothing helped. I also wrote a simple test to check if my sd card read speeds are too low for 192kbps or 320 kbps mp3, however tested on 1MB txt file and 20MHz freq average read speed is ~600kbps. Turning off logging info also doesnt change anything.

### Device Description

Raspberry pi pico 2 w with pcm5102 dac and generic cd card reader

### Sketch

```cpp
/**
 * @file player-sdfat-i2s.ino
 * @brief see https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sdfat-i2s/README.md
 * 
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext,17,20);
I2SStream i2s;
MP3DecoderHelix decoder;
AudioPlayer player(source, i2s, decoder);

void printMetaData(MetaDataType type, const char* str, int len){
  Serial.print("==> ");
  Serial.print(toStr(type));
  Serial.print(": ");
  Serial.println(str);
}

void setup() {
  Serial.begin(115200);
  //AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);

  // setup output
  auto config = i2s.defaultConfig(TX_MODE);
  config.pin_ws = 10;
  config.pin_bck = 11;
  config.pin_data = 9;
  i2s.begin(config);

  // setup player
  //source.setFileFilter("*Bob Dylan*");
  player.setMetadataCallback(printMetaData);
  player.setBufferSize(2048);
  player.begin();
}

void loop() {
  player.copy();
}
```

### What is your development environment

Arduino Ide

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

trybula, 2025-01-25:
Maybe it is also worth mentioning that i tried reencoding mp3 files but still nothing changed ;)

pschatzmann, 2025-01-25:
Did you try to increase the i2s buffer ? 

I would make it big enough to hold 2 - 3 decoded writes.

I also think your read speed is a bit low compared to what [we can get via Wifi](https://www.pschatzmann.ch/home/2023/02/28/arduino-audiotools-esp32-vs-rp2040-pico-w/). What SPI frequency did you try ?

pschatzmann, 2025-01-25:
Also google the topic: I e.g. [found this](https://forum.arduino.cc/t/slow-microsd-write-speeds-on-rp2040-pico-boards/952459/3)

pschatzmann, 2025-01-25:
Since this is not related to my functionality, I am converting this to a discussion!