pschatzmann/arduino-audio-tools #1888: CodecFLACFoxen.h crashes my esp32 when the end() function is called.

UnicronNL, 2025-01-30:
### Problem Description

When calling the `end()` function of AudioPlayer.h my esp32 panics, I think it has to do with the `end()` function in `CodecFLACFoxen.h`  
 ``` 
  void end() {
    TRACEI();
    flush();
    if (flac != nullptr) {
      foxen_data.resize(0);
      flac = nullptr;
    }
    buffer.resize(0);
    out.resize(0);
    is_active = false;
  }  
```
The ```flush();``` is called all the time, but if flac == nullptr it panics.  moving the ```flush();``` to the ```if (flac != nullptr)``` part the functionallity seems correct.
```
  void end() {
    TRACEI();
    if (flac != nullptr) {
      flush();
      foxen_data.resize(0);
      flac = nullptr;
    }
    buffer.resize(0);
    out.resize(0);
    is_active = false;
  }
```
So I think it is a bug, but please correct me if I am wrong.

### Device Description

ESP32 Audio Kit

### Sketch

```cpp
void end() {
    TRACEI();
    flush();
    if (flac != nullptr) {
      foxen_data.resize(0);
      flac = nullptr;
    }
    buffer.resize(0);
    out.resize(0);
    is_active = false;
  }
```

### What is your development environment

Arduino

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-01-30:
I committed some corrections

closed 2025-01-30 by pschatzmann

UnicronNL, 2025-01-30:
Great thank you!