pschatzmann/arduino-audio-tools #1906: esp32-c3

weirdorr, 2025-02-08:
### Problem Description

In file included from src/main.cpp:5:
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h: In constructor 'audio_tools::MP3DecoderHelix::MP3DecoderHelix()':
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h:25:22: error: 'class libhelix::MP3DecoderHelix' has no member named 'setReference'
                 mp3->setReference(this);
                      ^~~~~~~~~~~~
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h: In constructor 'audio_tools::MP3DecoderHelix::MP3DecoderHelix(Print&)':
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h:40:22: error: 'class libhelix::MP3DecoderHelix' has no member named 'setReference'
                 mp3->setReference(this);
                      ^~~~~~~~~~~~
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h: In constructor 'audio_tools::MP3DecoderHelix::MP3DecoderHelix(Print&, audio_tools::AudioInfoSupport&)':
.pio/libdeps/esp32-c3/audio-tools/src/AudioCodecs/CodecMP3Helix.h:59:22: error: 'class libhelix::MP3DecoderHelix' has no member named 'setReference'
                 mp3->setReference(this);
                      ^~~~~~~~~~~~

### Device Description

error

### Sketch

```cpp
#include <Arduino.h>
#include <LittleFS.h>
#include "AudioTools.h"
#include "AudioLibs/AudioSourceLittleFS.h"
#include "AudioCodecs/CodecMP3Helix.h"
#include "MyButton.h"
#include "MyLed.h"
```

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-02-08:
https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working

closed 2025-02-08 by pschatzmann

mikalhart, 2025-03-04:
For what it's worth, I, too get this exact error when simply compiling the latest **player-sdfat-i2s** example with "ESP32C3 Dev Module" (or any board, really) using the latest 2.3.4 Arduino. I notice that CodecMP3Helix.h has changed recently. The old version (which _did_ compile correctly) called

`            filter.setDecoder(mp3);
`

whereas the new one does this:

`            mp3->setReference(this);
`

I shall study the problem some more, but the provided `libhelix::MP3DecoderHelix` definitely doesn't provider a `setReference` method in any of its ancestors. (Was this class intended to inherit from `BinaryContainer`? That DOES implement `setReference`).

pschatzmann, 2025-03-04:
I think you just forgot to install the latest version of libhelix which [provides this method](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)!

mikalhart, 2025-03-04:
Oof. Indeed. Thank you.