pschatzmann/arduino-audio-tools #1916: RP2040 SPDIFOut error: "pins bck: '-1' and ws: '-1' must be next to each other"

Sola85, 2025-02-15:
### Problem Description

I compiled `streams-generator-spdif.ino` for ESP32 and it works great. I now wanted to move to an RP2040 (because I was hoping to get SPDIF rx and tx to work at the same time using [this](https://github.com/elehobica/pico_spdif_rx/)).

But even standalone `streams-generator-spdif.ino` doesn't seem to work. 
It compiles, but I see `SPDIFOut.begin()` generate an error message:
```
[E] I2SRP2040.h : 83 - pins bck: '-1' and ws: '-1' must be next to each other
```

This is followed by
```
19:40:27.631 -> [I] StreamCopy.h : 420 - try write  - 18 (open 1024 bytes) 
19:40:27.673 -> [I] StreamCopy.h : 420 - try write  - 19 (open 1024 bytes) 
19:40:27.673 -> [I] StreamCopy.h : 420 - try write  - 20 (open 1024 bytes) 
19:40:27.673 -> [I] StreamCopy.h : 420 - try write  - 21 (open 1024 bytes) 
19:40:27.673 -> [E] StreamCopy.h : 413 - write  to target has failed after 22 retries! (1024 bytes)
```

### Device Description

Original Raspberry Pi Pico

### Sketch

```cpp
Unmodified `streams-generator-spdif.ino`.

In order to see the above error message, I added 

bool res = out.begin(config);
while (!res){
  Serial.println("Failed to set up I2S");
  delay(1000);
  res = out.begin(config);
}

because I had trouble catching the begin of the logs.
```

### What is your development environment

Windows 11, Arduino 2.3.4, [Raspberry Pi Pico Arduino core 4.4.3](https://github.com/earlephilhower/arduino-pico/) 

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-02-15:
I doubt that the RP2040 is powerfull enough, but feel free to submit a Pull Request with the correction if you manage to make it work.
Just have a look at the implementation of the SPDIFOutput class

closed 2025-02-15 by pschatzmann

Sola85, 2025-02-15:
Thanks for the response. 
Just to be clear: the unmodified example code for `SPDIFOut` compiles but does not work on the rp2040. Is this intended?
(Just trying to understand why the issue is closed.)

pschatzmann, 2025-02-15:
Why don't you try to fix it yourself ? After all the error message tells you what needs to be done...

Sola85, 2025-02-16:
> I don't have any means to test any correction

I see, no worries.

I was able to work around the issue by overwriting the `begin` method of `SPDIFOutput` to use two "dummy pins" for ws and bck.

```cpp
class RP2040_SPDIFOutput: public SPDIFOutput {
  public:
    bool begin(SPDIFConfig config) {
      ...
      i2s_cfg.pin_ws = cfg.pin_data-2;  //Was previously i2s_cfg.pin_ws = -1;
      i2s_cfg.pin_bck = cfg.pin_data-1; //Was previously i2s_cfg.pin_bck = -1;
      i2s_cfg.pin_data = cfg.pin_data;
      ...
    }
};
```

I also had to increase the RP2040 clock to 192MHz using `set_sys_clock_khz(192000, false);`, but now SPDIF output works.

pschatzmann, 2025-02-17:
Based on your comments, I have added [some additional config parameters](https://github.com/pschatzmann/arduino-audio-tools/commit/da5fb4b439add6cb8d1d2860688627f2ae2d9880) for the RP2040