pschatzmann/arduino-audio-tools #1933: I'm having problems playing online MP3 files.

coco1667, 2025-02-26:
### Problem Description

Dear author, I'm very sorry to bother you. I'd like to ask you a question. I'm using an ESP32 as my hardware device. I want to play MP3 audio files from the Internet and I'm trying to achieve this by using your example code in examples/examples-communication/http-client/player-url-i2s/player-url-i2s.ino.
The serial monitor can monitor the data normally, but unfortunately, there is no sound. I'm sure there is nothing wrong with my hardware. The audio amplifier and the speaker can play sound. I'm really confused about this. Do you know the reason? Thank you very much.

### Device Description

ESP32-S3

### Sketch

```cpp
examples/examples-communication/http-client/player-url-i2s/player-url-i2s.ino.
```

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-02-26:
Please read this [wiki](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)!

closed 2025-02-26 by pschatzmann

coco1667, 2025-04-06:
Thank you. I've already found out what the problem is. It might be due to hardware issues in general. Thank you for your support and help. I wish you a happy life!

coco1667, 2025-04-06:
> 请阅读此[维基](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)！

Thank you. I've already found out what the problem is. It might be due to hardware issues in general. Thank you for your support and help. I wish you a happy life!