pschatzmann/arduino-audio-tools #1973: fatal error: esp_crt_bundle.h: No such file or directory     7 | #include "esp_crt_bundle.h"       |

Admsher, 2025-03-23:
### Problem Description

I am trying to use the sudio-tools library in esp-idf v5.3 and i am stuck with the error that esp_crt_bundle.h is missing for /audio-tools/src/AudioTools/CoreAudio/AudioHttp/URLStreamESP32.h

To clarify I have  checked this issue on esp-idf : https://github.com/espressif/esp-idf/issues/10464 and the flag is enabled in my case.

The file structure is that audio-tools,esp32-a2dp are present in components directory with arduino.

### Device Description

ESP32 Chip (as given in ESP-IDF). The board is ESP32 One

### Sketch

```cpp
There is no sketch but error is given as follows:

/home/admsher/esp32-new/components/audio-tools/src/AudioTools/CoreAudio/AudioHttp/URLStreamESP32.h:7:10: fatal error: esp_crt_bundle.h: No such file or directory
    7 | #include "esp_crt_bundle.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Other Steps to Reproduce

Add audiotools and esp32-a2dp in cmakelists as esp-idf components, also arduino is added as another component. 
Bluetooth 
```
[*] Bluetooth
        Host (Bluedroid - Dual-mode)  --->
        Controller (Enabled)  --->
    Bluedroid Options  --->
    Controller Options  --->
    Common Options  --->
[ ] Enable Bluetooth HCI debug mode

```

and
 ESP-TLS is enabled with Server and Client connections:
```
  Use Secure Element (ATECC608A) with ESP-TLS
[*] Enable client session tickets
[*] Enable server session tickets
(86400) Server session ticket timeout in seconds (NEW)
[ ] Certificate selection hook
[*] ESP-TLS Server: Set minimum Certificate Verification mode to Optional
[ ] Enable PSK verification
[ ] Allow potentially insecure options
```

### What is your development environment (incl. core version info)

ESP-IDF v5.3

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-23:
Google found this: https://github.com/espressif/esp-idf/issues/10464

closed 2025-03-23 by pschatzmann

Admsher, 2025-03-23:
Yes and I have gone through this and the flag is indeed enabled or 'y' in this case, but then I am not sure why it gives the same error for the audio-tools

Admsher, 2025-03-23:
😃 Great