pschatzmann/arduino-audio-tools #1974: error: deleting object of polymorphic class type 'audio_tools::Stream' which has non-virtual destructor might cause undefined behavior [-Werror=delete-non-virtual-dtor]   274 |       delete p_stream;

Admsher, 2025-03-23:
### Problem Description

The issue comes while compiling the library for  esp-idf v5.3 in the line 

```
void cleanupConverter() {
    if (p_stream != nullptr) {
      delete p_stream;
      p_stream = nullptr;
    }
  } 
```
for the file audio-tools/src/AudioTools/CoreAudio/AudioStreamsConverter.h:274 . 

For reference  p_stream is defined in 
"AudioTools/CoreAudio/AudioStreams.h:621" with the declaration 

```
 void setStream(Stream &stream) {
    TRACEI();
    p_stream = &stream;
    p_out = &stream;
  }
```

### Device Description

ESP32 Chip (As given by ESP-IDF)

### Sketch

```cpp
Provided int he issue explanation, no code written by user.
```

### Other Steps to Reproduce

If all things are okay with the other files , you should be prompted with this.

### What is your development environment (incl. core version info)

esp-idf v5.3 

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-23:
Check the log level of the message: I think this is only a warning!

Admsher, 2025-03-23:
It shows an error which then fails to build, I think it differs from esp-idf version to version

pschatzmann, 2025-03-23:
Fee free to submit a pull request

pschatzmann, 2025-03-23:
I mereged a pull request that should resolve this issue

Admsher, 2025-03-23:
Thanks a lot will try it out 👍

pschatzmann, 2025-03-23:
The IDF build was indeed broken: I committed some more corrections and it should now work again.

Admsher, 2025-03-24:
This issue is resolved now 👍

closed 2025-03-24 by Admsher