pschatzmann/arduino-audio-tools #1977: esp_bt_controller-init failed

Admsher, 2025-03-25:
### Problem Description

I was trying out the SDA2DP example on esp-idf v 5.3 and am stuck with an error in which the Bluetooth init is first acknowledged and then it says init failed.

### Device Description

ES32 One: https://www.waveshare.com/wiki/ESP32_One

### Sketch

```cpp
#define HELIX_LOGGING_ACTIVE false
#include "AudioTools.h"
#include "AudioTools/AudioLibs/A2DPStream.h"
#include "AudioTools/AudioLibs/AudioSourceSDFAT.h"
#include "AudioTools/AudioCodecs/CodecMP3Helix.h"

const char *startFilePath="/music/";
const char* ext="mp3";
AudioSourceSDFAT source(startFilePath, ext, 15);
A2DPStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

void setup() {
  Serial.begin(115200);
  AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Warning);
  // setup player
  // Setting up SPI if necessary with the right SD pins by calling 
  SPI.begin(14, 12, 13, 15);
  player.setVolume(0.8);
  player.begin();

  // setup output - We send the test signal via A2DP - so we conect to a Bluetooth Speaker
  auto cfg = out.defaultConfig(TX_MODE);
  cfg.silence_on_nodata = true; // prevent disconnect when there is no audio data
  cfg.name = "CLIP 5";  // set the device here. Otherwise the first available device is used for output
  //cfg.auto_reconnect = true;  // if this is use we just quickly connect to the last device ignoring cfg.name
  out.begin(cfg);
}

void loop() {
  player.copy();
}
```

### Other Steps to Reproduce

enable Bluetooth and BLE in esp-idf  which is needed to compile it in the first place. Strangely it works on Arduino IDE

### What is your development environment (incl. core version info)

ESP-IDF v5.3

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-25:
Did you [follow the instructions](https://github.com/pschatzmann/ESP32-A2DP/wiki/Espressif-IDF-as-a-Component) ?
Please share the log!

closed 2025-03-25 by pschatzmann

Admsher, 2025-03-25:
I have used arduino-esp32 for the same to include ArduinoSDFAT.h  and SDFat in my workspace . I hope it makes sense why these two are in the environment. I also checked the flags and they were enabled in my build.

pschatzmann, 2025-03-25:
I have never looked into the Arduino as component scenario, so I can't really help...