pschatzmann/arduino-audio-tools #1991: error: 'struct audio_driver::I2SDefinition' has no member named 'singal_type'

SaKiEQ, 2025-04-02:
### Problem Description

arduino-audio-tools-main/src/AudioTools/AudioLibs/I2SCodecStream.h:306:19: error: 'struct audio_driver::I2SDefinition' has no member named 'singal_type'
  306 |     codec_cfg.i2s.singal_type = (singal_t) info.signal_type;
      |                   ^~~~~~~~~~~

AudioTools/AudioLibs/I2SCodecStream.h:306:34: error: 'singal_t' was not declared in this scope; did you mean 'signal'?
  306 |     codec_cfg.i2s.singal_type = (singal_t) info.signal_type;
      |                                  ^~~~~~~~
      |                                  signal

### Device Description

ESP32 /Arduino IDE 2

### Sketch

```cpp
Compilation Error, due to above error. Removing the offending line of code codec_cfg.i2s.singal_type = (signal_t) info.singal_type; solves the issue, fixing the typo signal -> signal does not solve the problem.
```

### What is your development environment (incl. core version info)

Arduino IDE 2.3.5 macOS

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-04-02:
What Hardware are you using this with ?
singal_t is defined in the arduino-audio-driver project

I can't reproduce your issue, so I am closing this: Make sure that you have the latest versions of all dependencies installed

closed 2025-04-02 by pschatzmann

SaKiEQ, 2025-04-02:
Weird, for me its a clear typo, which is sort of confirmed by the compiler
error.

   ' codec_cfg.i2s.singal_type = (sigal_t) info.sigal_type;'

For me this is supposed to be signal_type.

In terms of update, the only dependency that is is not yet fully latest
version, is the ESP32 board version where I am lagging 1 version
behind.

Let me try and update that and check again.

joba-1, 2025-04-02:
agreed, twice switched n and g in signal, 100% a bug

joba-1, 2025-04-03:
If you want, you can try [this fork](https://github.com/joba-1/arduino-audio-tools.git) (only change is the fix). Error was introduced 3 months ago.

I'll do a PR shortly...

pschatzmann, 2025-04-03:
You use an old version of the arduino-audio-driver (that did not provide the codec_cfg.i2s.singal_type) with an actual version of the arduino-audio-tools which expects this!

I have corrected the misspelling (in both projects), but this means that you need to upgrade both libraries!