pschatzmann/arduino-audio-tools #1994: FadeStream writeEnd() isn't called anywhere?

cdanis, 2025-04-03:
### Problem Description

I'm experiencing popping sounds at the end of a short WAV file being played via an AudioPlayer object.

The file in question ([attached](https://github.com/user-attachments/files/19591637/bug.zip)) does indeed end with a DC offset, but I expected the LastSampleFader used by FadeConverter to take care of that.

However, it looks like the [writeEnd](https://github.com/pschatzmann/arduino-audio-tools/blob/c91d75ef6b1cf1d5a676fcfebdf20aa665610691/src/AudioTools/CoreAudio/Fade.h#L315C8-L315C16)() method which implements the fade-to-silence on input EOF is never actually called anywhere in the source code?  (The AudioPlayer method writeEnd() does have calls, but that's different, and not applicable in this case.)

### Device Description

ESP32-S3 dev board (FeatherS3 by UM)

### Sketch

```cpp
will provide if needed
```

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-04-04:
Works as designed: The fading is used in the setActive() (play()/stop()).

It is conceptually not possible to fade something out after having processed all audio data. You will need to make sure that you play back files that fade out the recorded data properly.

closed 2025-04-04 by pschatzmann

cdanis, 2025-04-04:
> Works as designed: The fading is used in the setActive() (play()/stop()).
> 
> It is conceptually not possible to fade something out after having processed all audio data. You will need to make sure that you play back files that fade out the recorded data properly. Alternatively, if you know the file size, you can use the above functions...

Okay, but then what's the purpose of FadeStream::writeEnd(), and of LastSampleFader?  It's over 100 lines, almost a full third of Fade.h.  LastSampleFader gets called to look all the data that passes through any FadeStream, but then no other part of the library or examples use it in any way.

pschatzmann, 2025-04-04:
Interesting: So if you know that it is the last copy step you could call writeEnd just before the copy()