pschatzmann/arduino-audio-tools #1998: Compiling for Arduino Uno R3 fails

mawidme, 2025-04-04:
### Problem Description

When I try to compile "player-sdfat-vs1053.ino" from the examples (or any example in fact) for the Arduino Uno R3, it fails with the error below because the "AudioToolsLogger" class is not found. This makes me wonder whether the original Arduino Uno is supported by the library at all?
`'AudioToolsLogger' was not declared in this scope`

Furthermore, when I comment the line with AudioToolsLogger and try to compile, I get memory exceeded errors. Would I have to optimize this myself or are there any existing switches (besides e.g. LOG_NO_MSG=true) to reduce memory consumption?
`Error during build: text section exceeds available space in board`

### Device Description

Arduino Uno R3

### Sketch

```cpp
Compile vs1053 example from repo: examples/examples-vs1053/player-sdfat-vs1053/player-sdfat-vs1053.ino
```

### What is your development environment (incl. core version info)

Arduino IDE 2.3.5

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

mawidme, 2025-04-04:
Here are the exact memory exceeded messages:
```ketch uses 77536 bytes (240%) of program storage space. Maximum is 32256 bytes.
Global variables use 7415 bytes (362%) of dynamic memory, leaving -5367 bytes for local variables. Maximum is 2048 bytes.```

pschatzmann, 2025-04-04:
The Arduino UNO just has not enough RAM to use the AudioTools: to make it compile, I have deactivated the logger, but as you can see, even with this you run out of memory.

You can try to use some vs1053 and SD library only and just copy the data from your files to the vs1053

The UNO R4, which has much more RAM might be a slightly better choice if you want to stick with Arduino boards.
However, I recommend to use some RP2040/RP2050 or ESP32 together with an [I2S DAC](https://github.com/pschatzmann/arduino-audio-tools/wiki/External-DAC)...

closed 2025-04-04 by pschatzmann