pschatzmann/arduino-audio-tools #2024: Heap corruption with Vorbis decoder

UnforeseenOcean, 2025-04-27:
### Problem Description

`CodecVorbis.h:114 - copy: 0 - N/A`, `assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)`

Issue occurs when `end();` is called after playing audio

Please refer to image and code for more info

[image]

[image]

[ESP32-BT-Speaker.zip](https://github.com/user-attachments/files/19929284/ESP32-BT-Speaker.zip)

### Device Description

ESP32 AudioKit (ES8388) V1

### Sketch

```cpp
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"
#include "AudioTools/AudioCodecs/CodecVorbis.h"
#include "BluetoothA2DPSink.h"
#include "newsound_trimmed.h"

AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);
VorbisDecoder vorbis;

void playSound(int index) {
  const uint8_t* data = nullptr;
  uint32_t data_len = 0;
  switch (index) {
    case 0: {
      data = battlow;
      data_len = battlow_len;
      break;
    }
    case 1: {
      data = connected;
      data_len = connected_len;
      break;
    }
    default: {
      data = poweroff;
      data_len = poweroff_len;
      break;
    }
  }
  MemoryStream ogg(data, data_len);
  vorbis.setInput(ogg);
  vorbis.setOutput(kit);
  vorbis.begin();
  while (vorbis.copy()){};
  vorbis.end(); // Causes fault
}

void setup() {
  Serial.begin(115200);
  setupKeys();
  vorbis.setOutput(kit);
  a2dp_sink.start("ESP-BT-SPK");
  playSound(7);
}

void loop() {
  delay(10);
}
```

### Other Steps to Reproduce

Please use the attached sketch archive, core version is set to most compatible version I can find.

OS: Windows 11 23H2 x64

### What is your development environment (incl. core version info)

Arduino IDE 2.3.6/1.8.19, Espressif ESP32 core for Arduino 2.0.17

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-04-27:
Your allocating the MemoryStream ogg on the stack which means that it does not exist any more when the method is left!
Define it as global object l

closed 2025-04-27 by pschatzmann

UnforeseenOcean, 2025-04-27:
Then you'll have to answer the question I asked: How do you give it data that I can change if desired and not a single fixed thing I defined once as a global object?

pschatzmann, 2025-04-27:
Just check the class documentation: there is certainly a setter method!

UnforeseenOcean, 2025-04-27:
Okay, I admit I made a boo-boo while trying to read the functions. However all it changed was that the code became slightly less messy. It still crashes, and commenting out `vorbis.end();` causes Bluetooth functions to cease functioning (then the audio does not cause crashes). Could you take a look at the code as it is right now, and see if it's implemented correctly? Please, *provide some kind of example code snippet.*

```cpp
#include <OneButton.h>
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h" // install https://github.com/pschatzmann/arduino-audio-driver
#include "AudioTools/AudioCodecs/CodecVorbis.h"
#include "BluetoothA2DPSink.h"
#include "newsound_trimmed.h"

#define KEY_VOLUP 18
#define KEY_NEXT 23
#define KEY_PLAY 19
#define KEY_PREV 13
#define KEY_VOLDN 36
#define BATT_PIN 15

uint8_t curVolume = 63;
bool isMuted = false;
uint16_t battTick = 0;
uint16_t battCheckTick = 30000;
bool audioPlaying = false;

AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);
VorbisDecoder vorbis;
MemoryStream ogg;

OneButton volup_btn(KEY_VOLUP, true);
OneButton next_btn(KEY_NEXT, true);
OneButton play_btn(KEY_PLAY, true);
OneButton prev_btn(KEY_PREV, true);
OneButton voldn_btn(KEY_VOLDN, true);

void playSound(int index) {
  const uint8_t* data = nullptr;
  uint32_t data_len = 0;
  ESP_LOGI("SoundPlay", "Playing sound %d", index);
  switch (index) {
    case 0: {
      data = battlow;
      data_len = battlow_len;
      break;
    }
    case 1: {
      data = connected;
      data_len = connected_len;
      break;
    }
    case 2: {
      data = disconnected;
      data_len = disconnected_len;
      break;
    }
    case 3: {
      data = initSnd;
      data_len = initSnd_len;
      break;
    }
    case 4: {
      data = linkloss;
      data_len = linkloss_len;
      break;
    }
    case 5: {
      data = volmax;
      data_len = volmax_len;
      break;
    }
    case 6: {
[... 177 lines cut ...]
  ESP_LOGI("OneButton", "Key setup OK?");
}

void setup() {
  Serial.begin(115200);
  setupKeys();
  a2dp_sink.set_on_connection_state_changed(connectionStatusChanged);
  a2dp_sink.start("ESP-BT-DRG");
  playSound(7);
  ESP_LOGI("System", "Ready");
}

void loop() {
  volup_btn.tick();
  next_btn.tick();
  play_btn.tick();
  prev_btn.tick();
  voldn_btn.tick();
  battTick++;
  if (battTick >= battCheckTick) {
    checkBatteryLevel();
    battTick = 0;
  }
}
```

UnforeseenOcean, 2025-04-27:
The changed part is:
```cpp
AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);
VorbisDecoder vorbis;
MemoryStream ogg; // Here

(...)

void playSound(int index) {
(...)
  ogg.setValue(data, data_len); // And here
  ogg.begin();
  vorbis.setInput(ogg);
  vorbis.setOutput(kit);
  ESP_LOGW("SoundPlay", "Start play...");
  vorbis.begin();
  ESP_LOGW("SoundPlay", "Enter loop");
  while (vorbis.copy()) {
  };
  ESP_LOGW("SoundPlay", "Exit loop, terminating sound");
  vorbis.end(); // The code crashes the moment it enters this line
  ESP_LOGW("SoundPlay", "Play sound %d ok...", index);
}
```
Do please note, the sound *does* play to the end, but it still complains about something, along the lines of `[E] CodecVorbis.h : 114 - copy: 0 - N/A`.

pschatzmann, 2025-04-27:
I changed the code for VorbisDecoder so that it is not displaying any error message for the rc = 0

UnforeseenOcean, 2025-04-27:
Is the heap corruption caused by my code or was there some other cause? If it was my code, I would really appreciate you at least point at where in the code was incorrect.

pschatzmann, 2025-04-27:
No error for your side:  you can ignore the error message or install the latest version of the AudioTools library, if it is disturbing your...

UnforeseenOcean, 2025-04-27:
The issue isn't that the error message appears, it's that the program crashes or it has its core function rendered nonfunctional with nothing in-between. Try as I might, I can't figure out why `vorbis.end();` causes the crash, and why commenting it out renders Bluetooth useless (it does authenticate, but that's it).

pschatzmann, 2025-04-27:
If it is crashing, please provide the backtrace!

UnforeseenOcean, 2025-04-27:
I don't want to get pedantic, so here it is. It was also included as a decoded form in the first image.

```
[E] CodecVorbis.h : 114 - copy: 0 - N/A
[  3264][W][ESP32-BT-Speaker.ino:101] playSound(): [SoundPlay] Exit loop, terminating sound
CORRUPT HEAP: Bad head at 0x3ffef0c0. Expected 0xabba1234 got 0x3ffe4364

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40083dc5:0x3ffd1b70 0x40095461:0x3ffd1b90 0x4009aea5:0x3ffd1bb0 0x4009aaa1:0x3ffd1ce0 0x40084211:0x3ffd1d00 0x4009aed5:0x3ffd1d20 0x400e28ee:0x3ffd1d40 0x400e4dec:0x3ffd1d60 0x400d4981:0x3ffd1d80 0x401d549b:0x3ffd1db0 0x400e4e2e:0x3ffd1dd0 0x400d4981:0x3ffd1df0 0x400d9686:0x3ffd1e20 0x400daa0d:0x3ffd2020 0x400ebf0e:0x3ffd2050

ELF file SHA256: 3004873ae6de2123
```

UnforeseenOcean, 2025-04-28:
Additional info: Using a very short clip and this modified code, I got `assert failed: heap_caps_free heap_caps.c:381 (heap != NULL && "free() target pointer is outside heap areas")` error instead. Happens regardless of whether PSRAM is enabled or disabled.

```cpp
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"
#include "AudioTools/AudioCodecs/CodecVorbis.h"
#include "BluetoothA2DPSink.h"
#include "newsound_trimmed.h"
AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);
VorbisDecoder vorbis;
MemoryStream ogg;

void infoDump() {
  ESP_LOGI("Whodunit", "Flash size: %d, Sketch size: %d, PSRAM Size: %d, Free PSRAM: %d, Total Heap: %d, Free Heap: %d", ESP.getFlashChipSize(), ESP.getSketchSize(), ESP.getPsramSize(), ESP.getFreePsram(), ESP.getHeapSize(), ESP.getFreeHeap());
}
void logWhodunit(){
  ESP_LOGI("Whodunit", "Lowest Free Heap: %d, Total Heap: %d", ESP.getMinFreeHeap(), ESP.getHeapSize());
}

void playSound(int index) {
  const uint8_t* data = nullptr;
  uint32_t data_len = 0;
  ESP_LOGI("SoundPlay", "Playing sound %d", index);
  infoDump();
  logWhodunit();
  switch (index) {
    case 7: {
      data = volmin;
      data_len = volmin_len;
      break; 
    }
  }
  ESP_LOGW("SoundPlay", "Init data, len = %u", data_len);
  logWhodunit();
  ogg.setValue(data, data_len);
  logWhodunit();
  ogg.begin();
  logWhodunit();
  vorbis.setInput(ogg);
  vorbis.setOutput(kit);
  logWhodunit();
  ESP_LOGW("SoundPlay", "Start play...");
  vorbis.begin();
  logWhodunit();
  ESP_LOGW("SoundPlay", "Enter loop");
  while (vorbis.copy()) {
  };
  ESP_LOGW("SoundPlay", "Exit loop, destroy OGG Vorbis decoder...");
  logWhodunit();
  vorbis.end(); // <- This is where the program crashes
  
  ESP_LOGW("SoundPlay", "Freeing reserved memory for MemoryStream...");
  logWhodunit();
  ogg.end();
  
  ESP_LOGW("SoundPlay", "Play sound %d ok...", index);
}

void setup() {
  Serial.begin(115200);
  infoDump();
  logWhodunit();
  playSound(7);
}

void loop() {
}
```

pschatzmann, 2025-04-28:
I guess the issue is that the decoder automatically calls end() and calling a second end() in your code is leading to this crash.
I committed a correction that makes sure that  ov_clear() is not called when it has already been processed.

UnforeseenOcean, 2025-04-28:
```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x00000000  PS      : 0x00060530  A0      : 0x800d4b91  A1      : 0x3ffd22b0  
A2      : 0x3ffc87b8  A3      : 0x00000049  A4      : 0x3f4011b8  A5      : 0x00000023  
A6      : 0x3f402610  A7      : 0x3f4011cd  A8      : 0x800e611c  A9      : 0x3ffd2290  
A10     : 0x3ffc8770  A11     : 0x00000000  A12     : 0x00000020  A13     : 0x00000004  
A14     : 0x3ff50000  A15     : 0xff000000  SAR     : 0x00000012  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x40092dba  LEND    : 0x40092dc5  LCOUNT  : 0x00000000  

Backtrace: 0xfffffffd:0x3ffd22b0 0x400d4b8e:0x3ffd22d0 0x400d9a74:0x3ffd2300 0x400db092:0x3ffd2340 0x400edb6a:0x3ffd2370
```

[image]

pschatzmann, 2025-04-28:
Can you double check that you really have the latest correction ?

src/AudioTools/AudioCodecs/CodecVorbis.h line 44 has

    callbacks.close_func = nullptr;

I was just running a test on my desktop and could not reproduce any issues even when calling end multiple times.

UnforeseenOcean, 2025-04-28:
Yes it does. Please test the code as is, it seems like there is an edge case that isn't getting picked up by the test environment.

[image]

UnforeseenOcean, 2025-04-28:
For ease of debugging, here is the entire sketch folder as is.

[ESP32-BT-Speaker-v0.0.1.zip](https://github.com/user-attachments/files/19949868/ESP32-BT-Speaker-v0.0.1.zip)
audio-driver 0.1.3
audio-tools 1.1.1
arduino-libvorbis-tremor 0.2
ESP32-A2DP 1.8.7

pschatzmann, 2025-04-30:
You are obviously not using A2DPStream!
Please provide a simple sketch just using the CodecVorbis to demonstrate your issues!

UnforeseenOcean, 2025-04-30:
> You are obviously not using A2DPStream which is making sure that you can't concurrently write to the I2S output! Please provide a simple sketch just using the CodecVorbis to demonstrate your issues as [highlited here](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)!
A2DPStream object does not seem to support the AVRSP protocol, which is essential. Please confirm or deny this. As I said before, if support is confirmed, I will move to using A2DPStream.
And as to (probably, just guessing at this point) why it's not working on my end and it does on your side, it's probably because you are not using a cut-down version of *my* code, and instead are using a reimplementation of it without on-demand sound object creation and destruction, tied to the events happening within ESP32-A2DP library. **Please check using my code, not a rewrite.**

> ps. the AudioBoardStream explicitly provides button handling functionality, so I don't undertand why you are using your own library
This is because of its excellent delay-less button handling that doesn't require additional logic that I have to implement. No need to implement waits and press detection if I use that library.

pschatzmann, 2025-04-30:
Please check the A2DPStream documentation: the is using and exposing among others the BluetoothA2DPSink.

Here is an [example that shows how you can use the key handling   ](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sd-audiokit/player-sd-audiokit.ino) of the AudioBoardStream.

UnforeseenOcean, 2025-04-30:
Before that, here is the minimal, cut-down code demonstrating the issue. Please DO NOT change the parts or rewrite it. Test as is.

```cpp
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"
#include "AudioTools/AudioCodecs/CodecVorbis.h"
#include "BluetoothA2DPSink.h"
#include "newsound_trimmed.h"

AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);

void playSound() {
  const uint8_t* data = nullptr;
  uint32_t data_len = 0;
  
  MemoryStream ogg;
  VorbisDecoder vorbis;
  
  data = poweron;
  data_len = poweron_len;
  
  ESP_LOGW("SoundPlay", "Init data, len = %u", data_len);

  ogg.setValue(data, data_len);
  ogg.begin();
  vorbis.setInput(ogg);
  vorbis.setOutput(kit);
  ESP_LOGW("SoundPlay", "Start play...");
  vorbis.begin();
  ESP_LOGW("SoundPlay", "Enter loop");
  while (vorbis.copy()) {
    // delay(1);
  };
  ESP_LOGW("SoundPlay", "Exit loop, destroy OGG Vorbis decoder...");
  vorbis.end();
  ESP_LOGW("SoundPlay", "Freeing reserved memory for MemoryStream...");
  ogg.end();
  ESP_LOGW("SoundPlay", "Play sound %d ok...", index);
}

void connectionStatusChanged(esp_a2d_connection_state_t state, void *ptr) {
  switch (state) {
    case esp_a2d_connection_state_t::ESP_A2D_CONNECTION_STATE_DISCONNECTED:
      ESP_LOGI("Bluedroid", "A2DP Source Disconnected");
      playSound();
      break;
    case esp_a2d_connection_state_t::ESP_A2D_CONNECTION_STATE_CONNECTED:
    ESP_LOGI("Bluedroid", "A2DP Source Connected");
      playSound();
      break;
    default:
      // Nothing
      break;
  }
}

void setup() {
  Serial.begin(115200);
  a2dp_sink.set_on_connection_state_changed(connectionStatusChanged);
  a2dp_sink.start("ESP-BT-DRG");
  playSound();
  ESP_LOGI("System", "Ready");
}

void loop() {
}
```

pschatzmann, 2025-04-30:
Did you test your audio data with my test sketch ?

UnforeseenOcean, 2025-04-30:
What was the setting used to save the file? It seems like there is a buffer underflow bug when I use 32kbps OGG Vorbis audio, which all these files were encoded as.

UnforeseenOcean, 2025-04-30:
[image]

The software being used to prepare the input data is GoldWave 6.77.

pschatzmann, 2025-04-30:
Try with Audacity

UnforeseenOcean, 2025-04-30:
> Try with Audacity

Your settings, please? Was it a default setting?

pschatzmann, 2025-04-30:
How did you convert it to a .h file ?

UnforeseenOcean, 2025-04-30:
> How did you convert it to a .h file ?

HxD - Select all - Copy As - C, though `xxd` shows the same error when put into the code.

pschatzmann, 2025-04-30:
This works perfectly for me!

I suggest that you try to reinstall the main brach for 
- https://github.com/pschatzmann/arduino-libvorbis-tremor
- https://github.com/pschatzmann/arduino-libopus

UnforeseenOcean, 2025-04-30:
OK, will report back once those are reinstalled completely.

UnforeseenOcean, 2025-04-30:
Yes, it works correctly. Please update those two repos to include latest release.