pschatzmann/arduino-audio-tools #2027: Compile error with latest beta version (850392f)

UnforeseenOcean, 2025-04-28:
### Problem Description

It appears that the recent code change (commit `850392f`) was not implemented correctly. It relates to the change that fixes the typo `singal_type` and probably more.
```
In file included from c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/AudioBoardStream.h:4,
                 from C:\Users\Torbjorn\Desktop\ESP32-BT-Speaker\ESP32-BT-Speaker.ino:3:
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h: In member function 'audio_tools::I2SCodecConfig audio_tools::I2SCodecStream::defaultConfig(audio_tools::RxTxMode)':
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h:64:37: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'struct audio_tools::I2SCodecConfig' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
     memcpy(&cfg, &cfg1, sizeof(cfg1));
                                     ^
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h:24:8: note: 'struct audio_tools::I2SCodecConfig' declared here
 struct I2SCodecConfig : public I2SConfig {
        ^~~~~~~~~~~~~~
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h: In member function 'bool audio_tools::I2SCodecStream::beginCodec(audio_tools::I2SCodecConfig)':
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h:306:19: error: 'struct audio_driver::I2SDefinition' has no member named 'signal_type'; did you mean 'singal_type'?
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                   ^~~~~~~~~~~
                   singal_type
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h:306:34: error: 'signal_t' was not declared in this scope
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                                  ^~~~~~~~
c:\Users\Torbjorn\Documents\Arduino\libraries\audio-tools\src/AudioTools/AudioLibs/I2SCodecStream.h:306:34: note: suggested alternative: 'signal'
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                                  ^~~~~~~~
                                  signal
```

### Device Description

ESP32 AudioKit v1.0 (ES8388), though it's irrelevant since it fails to compile

### Sketch

```cpp
#include <OneButton.h>
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h" // install https://github.com/pschatzmann/arduino-audio-driver
#include "AudioTools/AudioCodecs/CodecVorbis.h"
#include "BluetoothA2DPSink.h"
#include "newsound_trimmed.h"

#define KEY_VOLUP 18
#define KEY_NEXT 23
#define KEY_PLAY 19
#define KEY_PREV 13
#define KEY_VOLDN 36
#define BATT_PIN 15

uint8_t curVolume = 63;
bool isMuted = false;
uint16_t battTick = 0;
uint16_t battCheckTick = 30000;

AudioBoardStream kit(AudioKitEs8388V1);
BluetoothA2DPSink a2dp_sink(kit);
VorbisDecoder vorbis;
MemoryStream ogg;

OneButton volup_btn(KEY_VOLUP, true);
OneButton next_btn(KEY_NEXT, true);
OneButton play_btn(KEY_PLAY, true);
OneButton prev_btn(KEY_PREV, true);
OneButton voldn_btn(KEY_VOLDN, true);

void infoDump() {
  ESP_LOGI("Whodunit", "Flash size: %d, Sketch size: %d, PSRAM Size: %d, Free PSRAM: %d, Total Heap: %d, Free Heap: %d", ESP.getFlashChipSize(), ESP.getSketchSize(), ESP.getPsramSize(), ESP.getFreePsram(), ESP.getHeapSize(), ESP.getFreeHeap());
}
void logWhodunit(){
  ESP_LOGI("Whodunit", "Lowest Free Heap: %d, Total Heap: %d", ESP.getMinFreeHeap(), ESP.getHeapSize());
}
[... 243 lines cut ...]
  playSound(7);
  ESP_LOGI("System", "Ready");
}

void loop() {
  volup_btn.tick();
  next_btn.tick();
  play_btn.tick();
  prev_btn.tick();
  voldn_btn.tick();
  battTick++;
  if (battTick >= battCheckTick) {
    checkBatteryLevel();
    battTick = 0;
  }
}
```

### What is your development environment (incl. core version info)

ESP32 2.0.17, Arduino IDE 2.3.6

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

UnforeseenOcean, 2025-04-28:
Actually, please wait a moment. I have a hunch.

pschatzmann, 2025-04-28:
Just make sure that you installed the latest audio-tools and audio-driver libraries.

UnforeseenOcean, 2025-04-28:
It is not completed, please check again. This is the compile error with the latest release.
```
C:\Users\Torbjorn\Documents\ArduinoOld\libraries\arduino-audio-tools-1.1.1\src/AudioTools/AudioLibs/I2SCodecStream.h:24:8: note: 'struct audio_tools::I2SCodecConfig' declared here
 struct I2SCodecConfig : public I2SConfig {
        ^~~~~~~~~~~~~~
C:\Users\Torbjorn\Documents\ArduinoOld\libraries\arduino-audio-tools-1.1.1\src/AudioTools/AudioLibs/I2SCodecStream.h: In member function 'bool audio_tools::I2SCodecStream::beginCodec(audio_tools::I2SCodecConfig)':
C:\Users\Torbjorn\Documents\ArduinoOld\libraries\arduino-audio-tools-1.1.1\src/AudioTools/AudioLibs/I2SCodecStream.h:306:19: error: 'struct audio_driver::I2SDefinition' has no member named 'signal_type'; did you mean 'singal_type'?
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                   ^~~~~~~~~~~
                   singal_type
C:\Users\Torbjorn\Documents\ArduinoOld\libraries\arduino-audio-tools-1.1.1\src/AudioTools/AudioLibs/I2SCodecStream.h:306:34: error: 'signal_t' was not declared in this scope
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                                  ^~~~~~~~
C:\Users\Torbjorn\Documents\ArduinoOld\libraries\arduino-audio-tools-1.1.1\src/AudioTools/AudioLibs/I2SCodecStream.h:306:34: note: suggested alternative: 'signal'
     codec_cfg.i2s.signal_type = (signal_t) info.signal_type;
                                  ^~~~~~~~
                                  signal

```

UnforeseenOcean, 2025-04-28:
That fixed the *compile* issue. Could you add a link to the audio driver library for others? (maybe with a note that says "please update both")