pschatzmann/arduino-audio-tools #2052: [GCC 12.2] [c++23] error: call of overloaded 'int24_3bytes_t(int)' is ambiguous

MindStudioOfficial, 2025-05-27:
### Problem Description

library fails to compile with latest toolchain for the ESP32 in platformio:

```
Compiling .pio\build\esp32_ota\FrameworkArduino\USB.cpp.o
In file included from .pio/libdeps/esp32_ota/audio-tools/src/AudioToolsConfig.h:882:
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h: In static member function 'static audio_tools::int24_3bytes_t std::numeric_limits<audio_tools::int24_3bytes_t>::lowest()':
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:140:97: error: call of overloaded 'int24_3bytes_t(int)' is ambiguous
  140 |        static audio_tools::int24_3bytes_t lowest() {return audio_tools::int24_3bytes_t(-0x7FFFFF);};
      |                                                                                     
            ^
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:42:3: note: candidate: 'audio_tools::int24_3bytes_t::int24_3bytes_t(float)'
Compiling .pio\build\esp32_ota\FrameworkArduino\USBCDC.cpp.o
   42 |   int24_3bytes_t(const float in) {
      |   ^~~~~~~~~~~~~~
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:38:3: note: candidate: 'audio_tools::int24_3bytes_t::int24_3bytes_t(const int64_t&)'
   38 |   int24_3bytes_t(const int64_t &in) {
      |   ^~~~~~~~~~~~~~
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:34:3: note: candidate: 'audio_tools::int24_3bytes_t::int24_3bytes_t(const int32_t&)'
   34 |   int24_3bytes_t(const int32_t &in) {
      |   ^~~~~~~~~~~~~~
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:28:3: note: candidate: 'audio_tools::int24_3bytes_t::int24_3bytes_t(const int16_t&)'
   28 |   int24_3bytes_t(const int16_t &in) {
      |   ^~~~~~~~~~~~~~
.pio/libdeps/esp32_ota/audio-tools/src/AudioTools/CoreAudio/AudioBasic/Int24_3bytes_t.h:16:7: note: candidate: 'constexpr audio_tools::int24_3bytes_t::int24_3bytes_t(const audio_tools::int24_3bytes_t&)'
   16 | class int24_3bytes_t  {
      |       ^~~~~~~~~~~~~~
[...]
```

here:

```cpp
#ifdef USE_TYPETRAITS
#include <limits>

namespace std {
    template<> class numeric_limits<audio_tools::int24_3bytes_t> {
    public:
       static audio_tools::int24_3bytes_t lowest() {return audio_tools::int24_3bytes_t(-0x7FFFFF);};
       static audio_tools::int24_3bytes_t min() {return audio_tools::int24_3bytes_t(-0x7FFFFF);};
       static audio_tools::int24_3bytes_t max() {return audio_tools::int24_3bytes_t(0x7FFFFF);};
    };
}

#endif
```
... and the same for int24_4_bytes

## Proposed fix

```cpp
namespace std {
    template<> class numeric_limits<audio_tools::int24_3bytes_t> {
    public:
       static audio_tools::int24_3bytes_t lowest() {return audio_tools::int24_3bytes_t(static_cast<int32_t>(-0x7FFFFF));};
       static audio_tools::int24_3bytes_t min() {return audio_tools::int24_3bytes_t(static_cast<int32_t>(-0x7FFFFF));};
       static audio_tools::int24_3bytes_t max() {return audio_tools::int24_3bytes_t(static_cast<int32_t>(0x7FFFFF));};
    };
}
```

and
```cpp
namespace std {
    template<> class numeric_limits<audio_tools::int24_4bytes_t> {
    public:
       static audio_tools::int24_4bytes_t lowest() {return audio_tools::int24_4bytes_t(static_cast<int32_t>(-0x7FFFFF));};
       static audio_tools::int24_4bytes_t min() {return audio_tools::int24_4bytes_t(static_cast<int32_t>(-0x7FFFFF));};
       static audio_tools::int24_4bytes_t max() {return audio_tools::int24_4bytes_t(static_cast<int32_t>(0x7FFFFF));};    };
}

```

### Device Description

- ESP32-WROOM-32

### Sketch

```cpp
Just included in platformio.ini not yet in code

lib_deps = 
	https://github.com/pschatzmann/arduino-audio-tools.git
```

### What is your development environment (incl. core version info)

```
[env:esp32dev]
platform = espressif32
platform_packages = 
	espressif/toolchain-xtensa-esp32@12.2.0+20230208
board = esp32dev
framework = arduino
[...]
build_unflags = -std=gnu++11
build_flags = -std=gnu++23
lib_compat_mode = strict
```

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-05-27:
Did you try to add -DUSE_INT24_FROM_INT to the build_flags ?

MindStudioOfficial, 2025-05-27:
> Did you try to add -DUSE_INT24_FROM_INT to the build_flags ?

Ok but shouldn't this be always enabled?

Otherwise the defined numeric_limits can't decide which constructor to call from an int literal.

pschatzmann, 2025-05-27:
Unfortunately not - on some environments int is defined as typedef int int32_t; Then this will lead to a a duplicate declaration compile errors

MindStudioOfficial, 2025-05-27:
I see, thanks for the quick response

closed 2025-06-15 by pschatzmann