pschatzmann/arduino-audio-tools #2054: ID3 parser drops non-English titles

vladkorotnev, 2025-05-30:
### Problem Description

There is a check here in the ID3 parser:

https://github.com/pschatzmann/arduino-audio-tools/blob/2d0fb718ead7405817692121dd57b364f3f4749b/src/AudioTools/CoreAudio/AudioMetaData/MetaDataID3.h#L457-L461

For English tags it's not a problem:

```
I (56845) FTPSource: Open: [/Music/00_Collection/!Music/!RUS-CIS/I am waiting for you last summer/Edge Party/03 Away from here.mp3]
I (56969) FTPStreamer: Streaming is starting
I (56969) FTPStreamer: Streamer did begin decoder
W (58457) FTPStreamer: Meta[Album] = [Edge Party]
W (58458) FTPStreamer: Meta[Artist] = [I am waiting for you last summer]
W (58459) FTPStreamer: Meta[Title] = [Away from here]
W (58465) FTPStreamer: Meta[Genre] = [Post-Rock]
```

For non-English this causes the track info to be not detected, because the tags are actually UTF-8:

```
I (181426) FTPSource: Open: [/Music/00_Collection/!Music/!JAP/!Ani/OreImo/～俺の妹がこんなに可愛いわけがないComplete Collection＋～俺妹コンプ＋！/1-04 白いココロ.mp3]
I (181547) FTPStreamer: Streaming is starting
I (181547) FTPStreamer: Streamer did begin decoder
W (181848) APL_MAIN: HEAP: 37047 free of 255739 (35651 minimum, 31744 contiguous available)
W (181849) APL_MAIN: PSRAM: 3364203 free of 4191019 (3336135 minimum, 3211264 contiguous available)
W (181857) APL_MAIN: Loop task stack water mark = 4384
[W] MetaDataID3.h : 460 - TAG TALB ignored
[W] MetaDataID3.h : 460 - TAG TPE1 ignored
[W] MetaDataID3.h : 460 - TAG TIT2 ignored
W (182922) FTPStreamer: Meta[Genre] = [Anime]
```

Suggestion: remove the check completely and let the user code handle whatever borked file they might run into. (LMK if that sounds good and I'll PR it)

More complex suggestion: perform a proper UTF8 validity check, but I'm not sure if the library has any code necessary for this already.

### Device Description

ESP32 + ESPerCDP rev001 board

### Sketch

```cpp
Not relevant, but provided just for posterity:

<summary>

#pragma once
#include <AudioTools.h>
#include <AudioTools/AudioCodecs/CodecHelix.h>

/// Size of compressed data buffer in PS-RAM
const size_t NET_DATA_BUFFER_SIZE = 256 * 1024;

class FTPStreamingPipeline {
public:
  std::function<void(MetaDataType, const std::string)> metadataCallback = nullptr;

  FTPStreamingPipeline(AudioStream * outputPort, float netDataReqPct = 90.0, int decoderCore = 0):
    netDataBorder(netDataReqPct),
    inPort(nullptr),
    outPort(outputPort),
    bufferNetData(NET_DATA_BUFFER_SIZE),
    queueNetData(bufferNetData),
    meta(codec),
    splitToMeta(),
    decoder(outPort, &meta),
    copierDownloading(queueNetData, *inPort),
    codecTask("NETDEC", 40000, 6, decoderCore),
    netTask("NETNET", 20000, 14, 1),
    copierDecoding(splitToMeta, queueNetData) {
        outPort = outputPort;
        semaNet = xSemaphoreCreateBinary();
        semaCodec = xSemaphoreCreateBinary();
        xSemaphoreGive(semaNet);
        xSemaphoreGive(semaCodec);

        codec.addDecoder(mp3, "audio/mpeg");
        codec.addDecoder(aac, "audio/aac");
        codec.addDecoder(wav, "audio/vnd.wave");

        splitToMeta.add(decoder);
        splitToMeta.add(outMeta);

        outMeta.setCallback(printMetaData);
        outMeta.begin();

        decoder.setDecoder(&meta);
        codec.setNotifyActive(true);
        codec.addNotifyAudioChange(*outPort);
        decoder.setNotifyActive(true);
        decoder.addNotifyAudioChange(*outPort);

        decoder.setOutput(*outPort);
        _that = this;
  }

  virtual ~FTPStreamingPipeline() {
    if(running) stop();

    vSemaphoreDelete(semaNet);
    vSemaphoreDelete(semaCodec);
    _that = nullptr;
  }

  void start(Stream * in) {
    setSource(in);
    running = true;
    queueNetData.begin();
    ESP_LOGI(LOG_TAG, "Streaming is starting");
    netTask.begin([this, in]() {                     
        startDecoding();
        netLoop();
      }
    );
  }

  void stop() {
    NullStream tmp;
    decoder.setOutput(tmp);
    codec.setOutput(tmp);
    codec.end();
    decoder.end();

    running = false;
    xSemaphoreTake(semaNet, pdMS_TO_TICKS(1000));
    ESP_LOGI(LOG_TAG, "Streamer finalizing net task");
    netTask.end();

    // Small shitshow because the codec thread sometimes gets stuck and never exits and leaves the semaphore
    ESP_LOGI(LOG_TAG, "Streamer finalizing codec task");
    bool codecNotStuck = xSemaphoreTake(semaCodec, pdMS_TO_TICKS(1000));
    codecTask.end();

    if(!codecNotStuck) { 
        delay(1000); // give it time to unshit itself to prevent explosions 
    }
  }

  int getNetBufferHealth() {
    if(!running) return 100;
    return trunc(bufferNetData.levelPercent());
  }

protected:
  bool running = false;
  Task netTask;

  void setSource(Stream * source) {
    inPort = source;
    copierDownloading.begin(queueNetData, *inPort);
  }

  void startDecoding() {
    decoder.setNotifyActive(true);
    decoder.begin();
    ESP_LOGI(LOG_TAG, "Streamer did begin decoder");

    TickType_t last_successful_copy = xTaskGetTickCount();
    TickType_t last_stats = xTaskGetTickCount();

    codecTask.begin([this]() { 
      while(bufferNetData.levelPercent() < netDataBorder && running) { 
          delay(1);
      }
      
      if(!running) {
          ESP_LOGI(LOG_TAG, "Finishing up codec task early");
          decoder.end();
          xSemaphoreGive(semaCodec);
          vTaskDelay(portMAX_DELAY);
      }

      while(running) {
          copierDecoding.copy();
          delay(1);
      }

      ESP_LOGI(LOG_TAG, "Finishing up codec task");
      decoder.end();
      xSemaphoreGive(semaCodec);
      vTaskDelay(portMAX_DELAY);
    });
  }

  void netLoop() {
    TickType_t last_successful_copy = xTaskGetTickCount();
    while(true) {
      xSemaphoreTake(semaNet, portMAX_DELAY);
      TickType_t now = xTaskGetTickCount();
      int copied = copierDownloading.copy();
      if(copied > 0) {
          last_successful_copy = now;
          delay(2);
      } else if(copied < 0) {
          ESP_LOGE(LOG_TAG, "copied = %i ???", copied);
      } else if(bufferNetData.isFull()) {
          ESP_LOGW(LOG_TAG, "Net buffer full, no space to receive more data!!");
          delay(10);
      } else {
          // copied nothing, maybe yield to another task
          delay(10);
      }

      if(!running) {
          xSemaphoreGive(semaNet);
          break;
      }

      xSemaphoreGive(semaNet);
    }
    ESP_LOGI(LOG_TAG, "Finishing up net task");
    vTaskDelay(portMAX_DELAY);
  }

  void processMetadata(MetaDataType kind, const std::string value) {
    ESP_LOGW(LOG_TAG, "Meta[%s] = [%s]", toStr(kind), value.c_str());
    if (metadataCallback) metadataCallback(kind, value);
  }
private:
  Stream * inPort;
  MP3DecoderHelix mp3;
  AACDecoderHelix aac;
  WAVDecoder wav;
  MultiDecoder codec;
  MetaDataFilterDecoder meta;
  MetaDataOutput outMeta;
  MultiOutput splitToMeta;
  StreamCopy copierDownloading;
  StreamCopy copierDecoding;
  EncodedAudioStream decoder;
  BufferRTOS<uint8_t> bufferNetData;
  QueueStream<uint8_t> queueNetData;
  AudioStream * outPort;
  SemaphoreHandle_t semaNet;
  SemaphoreHandle_t semaCodec;
  Task codecTask;
  float netDataBorder;
  static FTPStreamingPipeline * _that;

  static void printMetaData(MetaDataType type, const char* str, int len){
    if (_that) {
      _that->processMetadata(type, str);
    }
  }

  const char * LOG_TAG = "FTPStreamer";
};

FTPStreamingPipeline* FTPStreamingPipeline::_that = nullptr;

</summary>
```

### What is your development environment (incl. core version info)

```
PLATFORM: Espressif 32 (6.10.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (custom) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b 
 - framework-espidf @ 3.40407.240606 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
 Dependency Graph
|-- ESP32-A2DP @ 1.8.6+sha.e498b67
|-- audio-tools @ 1.1.1+sha.3f4b25a
|-- TinyFTPClient @ 1.2.0+sha.030d7a0
|-- libhelix @ 0.8.7+sha.a2b00be
|-- ESP32-targz @ 1.2.7
|-- HTTPClient @ 2.0.0
|-- ArduinoJson @ 7.3.1
|-- LittleFS @ 2.0.0
|-- ESPer-Core @ 1.0
|-- WiFi @ 2.0.0
|-- ESPer-GUI @ 1.0
|-- ArduinoOTA @ 2.0.0
|-- ESPer-CDP @ 1.0
 ```

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-05-30:
I had to add this check because w/o it the playback of a lot of files were just creashing the playback.
If you want to disable it I recommend to add a separate method to disable it on request or disable it via a preprocessor command

vladkorotnev, 2025-05-30:
Put a PR up, works for me when the flag is enabled in the config

[image]

PR #2056 (2025-05-30): Allow to disable ASCII check in ID3 parser
Fix the inability to show Japanese/Chinese/Russian metatags in MP3 files with ID3 metadata.

*Feature-flagged:* must `-DAUDIOTOOLS_ID3_TAG_ALLOW_NONASCII=true` to see any effect.

Closes #2054
closed 2025-05-30 by pschatzmann