pschatzmann/arduino-audio-tools #2069: setVolume using variable

peteDDD, 2025-06-10:
### Problem Description

Using `setVolume()` with a discrete value, for example, `setVolume(0.5);` works great.
Trying to pass a value to setVolume() is however not working for me.  It is as if the value ends up at zero even though volume.volume() returns the value I have for the variable I used.

six hours of trying to diagnose this with dozens of tests...   I imagine you will point out a simple error that I am simply blind to ...
PS: Many thanks for a great library!

### Device Description

ESP32-S3-DEVKIT-C1  N16R8
MAX98357A

### Sketch

```cpp
void mySound::initSound() {
    static const char* TAG = "initSound";
    pinMode(SOUND_SD_MODE_PIN, OUTPUT);
    digitalWrite(SOUND_SD_MODE_PIN, SOUND_SHUTDOWN);
    // Initialize our overriding I2S config file first with default settings
    myI2Sconfig = I2Sout.defaultConfig(TX_MODE);
    // Now configure the pins that are different than in the default config
    myI2Sconfig.pin_ws = SOUND_LRC_PIN;   // aka LRC   
    myI2Sconfig.pin_bck = SOUND_BCLK_PIN;  // aka BCLK  
    myI2Sconfig.pin_data = SOUND_DIN_PIN; // aka DIN   
    I2Sout.begin(myI2Sconfig);

    // set initial volume
    auto vcfg = volume.defaultConfig();
    vcfg.copyFrom(myI2Sconfig);
    volume.begin(vcfg);

    ESP_LOGI(TAG, "Jconfig.SOUND_VOLUME: %f", Jconfig.SOUND_VOLUME);

    volume.setVolume(Jconfig.SOUND_VOLUME); // DOES NOT WORK

    //volume.setVolume(static_cast<float>(Jconfig.SOUND_VOLUME));  // DOES NOT WORK

    //float volumeToUse = Jconfig.SOUND_VOLUME);  // These two lines DO NOT WORK
    //volume.setVolume(volumeToUse); 

    //volume.setVolume(0.5); // WORKS

    ESP_LOGI(TAG, "volume.volume(): %f", volume.volume());
}

The diagnostics output from this portion of the code:

[ 28742][I][mySound.cpp:38] initSound(): [initSound] Jconfig.SOUND_VOLUME: 0.500000
[ 28743][I][mySound.cpp:44] initSound(): [initSound] volume.volume(): 0.500000

Jconfig.SOUND_VOLUME is a float.   I confirmed that the hex value of Jconfig.SOUND_VOLUME exactly matches the intended float value.  I have tested with various values:  0.30, 0.50, 1.00
```

### Other Steps to Reproduce

Other supporting code:

```
void mySound::playWAV(const char* fileName) {
    static const char* TAG = "playWAV";
    if (!LittleFS.begin()) {
        ESP_LOGE(TAG, "LittleFS initialization failed!");
        return;
    }

    String fullPath = String(soundDir) + String(fileName);
    ESP_LOGD(TAG, "******Opening WAV file: %s", fullPath.c_str());
    audioFile = LittleFS.open(fullPath.c_str(), "r");
    b_AskedToPlayWav = true;
    I2Sout.begin(myI2Sconfig);
    decoder.begin();
    digitalWrite(SOUND_SD_MODE_PIN, SOUND_TURNON);
    copier.begin(decoder, audioFile);
}

bool mySound::wavIsDonePlaying() {
    if (!copier.copy()) {
        terminateWav();
        return true; // done playing wav
    }
    else {
        return false; // still playing wav
    }
}

void mySound::terminateWav() {  // to terminate Wav that is already playing (before starting a new Wav)
    audioFile.close();
    b_AskedToPlayWav = false; // done playing wave
    decoder.end();
    I2Sout.end();
    digitalWrite(SOUND_SD_MODE_PIN, SOUND_SHUTDOWN);
}

bool mySound::wavWasAskedToPlay() {
    return b_AskedToPlayWav;
}

void mySound::waitForWavToFinish() {
    while (!wavIsDonePlaying()) { // this must be called for the sound file to keep playing
    }
}
```

example that calls, plays, and cleans up

```
sound.initsound();
sound.playWAV("/Hello.wav"]);
sound.waitForWavToFinish();
```

### What is your development environment (incl. core version info)

[env]
platform = espressif32@6.5.0
board = esp32-s3-devkitc-1
framework = arduino
board_build.filesystem = LittleFS
;lib_ldf_mode = deep
lib_ldf_mode = chain+
lib_extra_dirs = ~/Documents/Arduino/libraries
monitor_speed = 115200
build_unflags = -std=gnu++11
build_flags = 
    -Wno-unused-variable ; Schatzmann
    -Wno-unused-but-set-variable ; Schatzmann
    -Wno-unused-function ; Schatzmann
    -Wno-format-extra-args ; Schatzmann
	-std=gnu++17
	-fmax-errors=5
	-I./src/
	-I./src/OTAdrive
	-I./src/esp32logger
	-DCORE_DEBUG_LEVEL=5
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
	-I"${platformio.packages_dir}/framework-arduinoespressif32/libraries/FS/src"
	-DI2C_BUFFER_LENGTH=255

lib_deps = 
	adafruit/Adafruit NeoPixel@^1.12.3
	vshymanskyy/TinyGSM@^0.11.5
	vshymanskyy/StreamDebugger@^1.0.1
	bblanchon/ArduinoJson@^6.21.1
	fbiego/ESP32Time@^2.0.6
	marian-craciunescu/ESP32Ping@^1.7
	https://github.com/pschatzmann/arduino-audio-tools.git

lib_ignore = 
	Time

[env:ESP32-S3-N16R8-BLACK-AI]
board = esp32-s3-devkitc-1
board_build.arduino.memory_type = qio_opi
board_build.flash_mode = qio
board_build.psram_type = opi
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
board_build.partitions = N16R8_partitions.csv  ; 2 x 7MB APP partitions, 1.875MB SPIFFS

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

peteDDD, 2025-06-11:
I completely agree with you Phil.  That's why I spent hours on this trying to prove to myself that the variable was not a float and doing everything I could to force floats in different ways.   Even the volume.volume() returned the correct value. (you can see that below the code).  But the sound is audible when I give set volume a constant and not when I give it a float variable... Even though it returns the same float value when I query with volume.volume.   I try more in the morning...

pschatzmann, 2025-06-11:
Did you try to reproduce your issue with the test example that I gave you ?

peteDDD, 2025-06-11:
Yes I did... and you know what... my code also shows that the volume was "set"... but still the sound only is audible if I use a discrete  value to set the volume.  This is so very strange...

again, part of my output

```
[I] VolumeStream.h : 199 - setVolume: 0.500000 at 0
[I] VolumeStream.h : 199 - setVolume: 0.500000 at 1
[ 26614][I][mySound.cpp:74] initSound(): [initSound] volume_readback: 1.000000 

[ 26614][I][mySound.cpp:77] initSound(): [initSound] volume.volume(): 0.500000
```

and the output from the test code you sent:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x1b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[    89][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[I] VolumeStream.h : 199 - setVolume: 1.000000 at 0
[I] VolumeStream.h : 199 - setVolume: 1.000000 at 1
[I] VolumeStream.h : 199 - setVolume: 0.100000 at 0
[I] VolumeStream.h : 199 - setVolume: 0.100000 at 1
[I] SoundGenerator.h : 167 - SineWaveGenerator::begin(channels=2, sample_rate=44100, frequency=493.88)
[I] SoundGenerator.h : 150 - bool audio_tools::SineWaveGenerator<T>::begin() [with T = short int]
[I] AudioTypes.h : 125 - SoundGenerator: sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] Buffers.h : 406 - resize: 4
[I] SoundGenerator.h : 194 - setFrequency: 493.88
[I] SoundGenerator.h : 195 - active: true
started...
0,0
44,44
89,89
134,134
177,177
220,220
262,262
302,302
341,341
378,378
414,414
447,447
478,478
507,507
533,533
556,556
577,577
595,595
610,610
622,622
631,631
637,637
639,639
639,639
635,635
628,628
```

pschatzmann, 2025-06-11:
So, this is working: your problem must be somewhere completely different.
You are setting up everything for Stereo, but use a Mono WAV.
Did you try with a stereo WAV ?

peteDDD, 2025-06-11:
Have not tried stereo   but since the sound plays is I setVolume one way but not the other, it is extraordinarily curious problem.

What I need to do is read up on setting it up for mono since this system is mono and I need to keep the files from getting too large.

peteDDD, 2025-06-11:
Seems something was wrong with the class library I created.  So, I started again clean on that.   It is now working.  Here is what I ended up with.  (and set to one channel ;^)

```
wavPlayer.h
#pragma once

#include <Arduino.h>
#include <LittleFS.h>
#include "AudioTools.h"
#include "AudioTools/AudioCodecs/CodecWAV.h"
#include "AudioTools/Disk/AudioSourceLittleFS.h"

class WavPlayer {
public:
    WavPlayer(uint8_t din_pin, uint8_t bclk_pin, uint8_t lrc_pin, uint8_t sd_mode_pin);
    
    void begin(float initial_volume = 1.0);
    void playWAV(const char* fileName);
    bool checkIfWavIsDonePlaying();
    void setVolume(float volume);
    bool getIsPlaying();

private:
    const uint8_t _din_pin;
    const uint8_t _bclk_pin;
    const uint8_t _lrc_pin;
    const uint8_t _sd_mode_pin;
    
    static const uint8_t SOUND_SHUTDOWN = 0;
    static const uint8_t SOUND_TURNON = 1;

    I2SStream _i2sOut;
    VolumeStream _volume;
    EncodedAudioStream _decoder;
    StreamCopy _copier;
    File _audioFile;
    bool _isPlaying;

    I2SConfig _i2sConfig;
    VolumeStreamConfig _volumeConfig;
}; 
```

wavPlayer.cpp
```
#include "WavPlayer.h"

WavPlayer::WavPlayer(uint8_t din_pin, uint8_t bclk_pin, uint8_t lrc_pin, uint8_t sd_mode_pin)
    : _din_pin(din_pin)
    , _bclk_pin(bclk_pin)
    , _lrc_pin(lrc_pin)
    , _sd_mode_pin(sd_mode_pin)
    , _volume(_i2sOut)
    , _decoder(&_volume, new WAVDecoder())
    , _isPlaying(false) {
}

void WavPlayer::begin(float initial_volume) {
    pinMode(_sd_mode_pin, OUTPUT);
    digitalWrite(_sd_mode_pin, SOUND_SHUTDOWN);

    AudioToolsLogger.begin(Serial, AudioToolsLogLevel::Info);

    if (!LittleFS.begin()) {
        Serial.println("LittleFS failed to mount");
        return;
    }

    _i2sConfig = _i2sOut.defaultConfig(TX_MODE);
    _i2sConfig.pin_ws = _lrc_pin;     // aka LRC  
    _i2sConfig.pin_bck = _bclk_pin;   // aka BCLK  
    _i2sConfig.pin_data = _din_pin;   // aka DIN  
    _i2sConfig.channels = 1;

    _volumeConfig = _volume.defaultConfig();
    _volumeConfig.copyFrom(_i2sConfig);

    _decoder.begin();
    _volume.setVolume(initial_volume);
}

void WavPlayer::playWAV(const char* fileName) {
    _audioFile = LittleFS.open(fileName, "r");
    _isPlaying = true;
    _i2sOut.begin(_i2sConfig);
    _decoder.begin();
    digitalWrite(_sd_mode_pin, SOUND_TURNON);
    _copier.begin(_decoder, _audioFile);
}

bool WavPlayer::checkIfWavIsDonePlaying() {
    if (!_copier.copy()) {
        terminateWav();
        return true; // done playing wav
    }
    return false; // still playing wav
}

void WavPlayer::terminateWav() {
    _audioFile.close();
    _isPlaying = false;
    _decoder.end();
    _i2sOut.end();
    digitalWrite(_sd_mode_pin, SOUND_SHUTDOWN);
}

void WavPlayer::setVolume(float volume) {
    if (volume > 1.0) {
        volume = 1.0;
    }
    if (volume < 0.0) {
        volume = 0.0;
    }
    _volume.setVolume(volume);

}

bool WavPlayer::getIsPlaying() {
    return _isPlaying;
}
```

Test program:
```
#include <Arduino.h>
#include "WavPlayer.h"

#define SOUND_DIN_PIN 41  // aka DATA
#define SOUND_BCLK_PIN 40  // aka BCK
#define SOUND_LRC_PIN 39  // aka WS
#define SOUND_SD_MODE_PIN 42  // aka shutdown_pin

// list of wav files to play, saved in data folder
const char* wavFiles[] = { "/earth_move.wav", "/tremble.wav", "/earth_move.wav" };
const int numOfWavFiles = sizeof(wavFiles) / sizeof(wavFiles[0]);
uint8_t currentWavFileIndex = 0;

WavPlayer wavPlayer(SOUND_DIN_PIN, SOUND_BCLK_PIN, SOUND_LRC_PIN, SOUND_SD_MODE_PIN);

float volumeFloat = 0.7;

void setup(void) {
  Serial.begin(115200);
  while (!Serial) {
    delay(100);
  }

  wavPlayer.begin();
  wavPlayer.setVolume(volumeFloat);
  wavPlayer.playWAV(wavFiles[currentWavFileIndex]);
}

void loop() {
  if (wavPlayer.getIsPlaying()) {
    if (wavPlayer.checkIfWavIsDonePlaying()) {
      if (currentWavFileIndex++ < numOfWavFiles - 1) {
        volumeFloat = volumeFloat - 0.2;
        wavPlayer.setVolume(volumeFloat);
        wavPlayer.playWAV(wavFiles[currentWavFileIndex]);
      }
    }
  }
}
```

closed 2025-06-11 by peteDDD