pschatzmann/arduino-audio-tools #2111: OggOpus Stream parse error

smokesea, 2025-07-26:
### Problem Description

I use OggOpus Encode pcm send to server but parse error。

$opusinfo test.opus

processing file "test.opus"...

New logical stream (#1, serial: f0320777): type opus
Encoded with Arduino
WARNING: Samples with negative granpos in stream 1
WARNING: Sample count ahead of granule (1920>978) in stream 1
WARNING: Sample count ahead of granule (2880>995) in stream 1
WARNING: Sample count ahead of granule (3840>1014) in stream 1
。。。。。
WARNING: Sample count ahead of granule (130560>3273) in stream 1
WARNING: EOS not set on stream 1 (normal for live streams)
Opus stream 1:
	ERROR: stream 1 has a negative duration: 2326--947-3840=-567
	ERROR: stream 1 has interior holes or more than one page of end trimming
	Pre-skip: 3840
	Playback gain: 0 dB
	Channels: 1
	Original sample rate: 16000 Hz
	Packet duration:   20.0ms (max),   20.0ms (avg),   20.0ms (min)
	Page duration:     20.0ms (max),   20.0ms (avg),   20.0ms (min)
	Total data length: 8350 bytes (overhead: 46.8%)
	Playback length: 0m:00.000s
	Average bitrate: 0 kbit/s, w/o overhead: 0 kbit/s

### Device Description

esp32s3 psram enabled. 

Platformio.ini
[env:seeed_xiao_esp32s3]
platform = espressif32
board = freenove_esp32_s3_wroom
framework = arduino
lib_deps = 
	fastled/FastLED@^3.10.1
	links2004/WebSockets@^2.6.1

### Sketch

```cpp
#include "audio_encode.h"
#include "AudioTools/AudioCodecs/CodecOpusOgg.h"
#include "AudioTools/Communication/WebSocketOutput.h"

uint8_t sample_buffer[BUFFER_SIZE];

AudioInfo encodeInfo(16000, 1, 16);
I2SStream mic;
WebSocketOutput wsOut(webSocket);
OpusOggEncoder opusOggEncoder;

uint8_t opus_buffer[MAX_PACKET_SIZE];  // Opus 编码输出

void EncodeTask(void *param){
    int encoder_error;
    
    // AudioLogger::instance().begin(Serial, AudioLogger::Debug); 
    //初始化MIC
    auto cfg = mic.defaultConfig(RX_MODE);
    cfg.setAudioInfo(encodeInfo);
    cfg.sample_rate = AUDIO_SAMPLE_RATE;
    cfg.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
    cfg.channels = I2S_CHANNEL_MONO;
    cfg.i2s_format = I2S_STD_FORMAT;
    cfg.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT;
    cfg.use_apll = false;
    cfg.auto_clear = true;
    cfg.is_master = true;
    cfg.fixed_mclk = 0;
    cfg.pin_ws = AUDIO_I2S_MIC_GPIO_WS;
    cfg.pin_bck = AUDIO_I2S_MIC_GPIO_SCK;
    cfg.pin_data_rx = AUDIO_I2S_MIC_GPIO_DIN;
    cfg.pin_data = I2S_PIN_NO_CHANGE;
    mic.begin(cfg);

    opusOggEncoder.setAudioInfo(encodeInfo);
    opusOggEncoder.setOutput(wsOut);
    opusOggEncoder.begin();

    while(1){
        int begin = millis();        
        int bytesRead = mic.readBytes(sample_buffer, BUFFER_SIZE);
        opusOggEncoder.write(sample_buffer, bytesRead);
        int delay_ms = millis() - begin;
        if (delay_ms < 20){
            vTaskDelay(20 - delay_ms);
        }
    }
}
```

### What is your development environment (incl. core version info)

PlatformIO Arduino Esp idf

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

smokesea, 2025-07-26:
other question: 

auto config = opusOggEncoder.config();
config.application = OPUS_APPLICATION_VOIP;
config.complexity = 0;
config.frame_sizes_ms_x2 = OPUS_FRAMESIZE_20_MS;

It seems that the above parameters are not applied to the Encoder.

smokesea, 2025-07-26:
how about this , It's been tormenting me for a long time.
$opusinfo test.opus

processing file "test.opus"...

New logical stream (https://github.com/pschatzmann/arduino-audio-tools/issues/1, serial: f0320777): type opus
Encoded with Arduino
WARNING: Samples with negative granpos in stream 1
WARNING: Sample count ahead of granule (1920>978) in stream 1

pschatzmann, 2025-07-26:
Most likely the bug in your copy logic: you don't check how much was effectively written and loose some data this way.
Use a StreamCopy copier to copy the data...

Please read the [The most important basics!](https://github.com/pschatzmann/arduino-audio-tools/wiki/Introduction)

smokesea, 2025-07-26:
I have changed to StreamCopy. The problem still exists. It's so wired. ffmpeg can also convert it successfully. but opusinfo can't.

`

#include "audio_encode.h"
#include "AudioTools/AudioCodecs/CodecOpusOgg.h"
#include "AudioTools/Communication/WebSocketOutput.h"

AudioInfo encodeInfo(AUDIO_SAMPLE_RATE, AUDIO_CHANNELS, 16);
I2SStream mic;
WebSocketOutput wsOut(webSocket);
OpusOggEncoder opusOggEncoder;
EncodedAudioStream audioEncoder(&wsOut, &opusOggEncoder);
StreamCopy copier(audioEncoder, mic);

void startAudioRecord(){
    auto& cfg = opusOggEncoder.config();
    cfg.setAudioInfo(encodeInfo);
    cfg.application = OPUS_APPLICATION_VOIP;
    cfg.bitrate = AUDIO_OPUS_BITRATE; //16000
    cfg.sample_rate = AUDIO_SAMPLE_RATE; //16000
    cfg.complexity = AUDIO_OPUS_COMPLEXITY; //0
    cfg.frame_sizes_ms_x2 = OPUS_FRAMESIZE_40_MS; //
    audioEncoder.begin(encodeInfo);
}

void EncodeTask(void *param){
    int encoder_error;
    
    // AudioLogger::instance().begin(Serial, AudioLogger::Debug); 
    auto cfg = mic.defaultConfig(RX_MODE);
    cfg.setAudioInfo(encodeInfo);
    cfg.sample_rate = AUDIO_SAMPLE_RATE;
    cfg.bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT;
    cfg.channels = I2S_CHANNEL_MONO;
    cfg.i2s_format = I2S_STD_FORMAT;
    cfg.channel_format = I2S_CHANNEL_FMT_ONLY_LEFT;
    cfg.use_apll = false;
    cfg.auto_clear = true;
    cfg.is_master = true;
    cfg.fixed_mclk = 0;
    cfg.pin_ws = AUDIO_I2S_MIC_GPIO_WS;
    cfg.pin_bck = AUDIO_I2S_MIC_GPIO_SCK;
    cfg.pin_data_rx = AUDIO_I2S_MIC_GPIO_DIN;
    cfg.pin_data = I2S_PIN_NO_CHANGE;
    mic.begin(cfg);

    startAudioRecord();

    while(1){
        if(!webSocket.isConnected()){
            vTaskDelay(10);
            continue;   
        }
        int begin = millis();
        copier.copy();
        int delay_ms = millis() - begin;
        if (delay_ms < 40){
            vTaskDelay(40 - delay_ms);
        }
    }
}
`