pschatzmann/arduino-audio-tools #2149: No audio output on Waveshare ESP32-S3-Touch-AMOLED-1.75 board using ES8311

dingetje, 2025-09-03:
### Problem Description

I'm not getting any audio output? Demo 08_ES8311 works but is using ESP_I2S library.

Expected: sine wave audio on audio output.

### Device Description

https://www.waveshare.com/wiki/ESP32-S3-Touch-AMOLED-1.75

### Sketch

```cpp
`
#include "../pins_config.h"
#include "AudioTools.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"

AudioInfo info(44100, 2, 16);
SineWaveGenerator<int16_t> sineWave(32000);     // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound(sineWave);  // Stream generated from sine wave
AudioBoardStream out(WaveShareBoard);
StreamCopy copier(out, sound);  // copies sound into i2s
`

In setup code:

`
  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.copyFrom(info);
  config.output_device = DAC_OUTPUT_ALL;
  config.pin_bck = BCLKPIN;
  config.pin_ws = WSPIN;
  config.pin_data = DOPIN;
  config.pin_mck = MCLKPIN;
  config.i2s_function = PinFunction::CODEC;
  config.i2s_format = I2S_STD_FORMAT;
  config.input_device = ADC_INPUT_NONE;
  config.sd_active = false;

  // cfg.i2s_port_no = 0; // or 1 if 0 is already in use
  if (!out.begin(config)) {
    Serial.println("I2S failed to start!");
    return false;
  }

  // additinal settings
  out.setVolume(0.8);
//  out.addHeadphoneDetectionAction();

  // start sine wave
  if (!sineWave.begin(info, N_B4))
  {
    Serial.println("SineWave begin failed!");
    return false;
  }
  Serial.println("started...");
`

In loop code:
`
void SoundLoop()
{
  copier.copy();
  out.processActions();
}
`
```
[... 187 lines cut ...]
````
[D] StreamCopy.h : 108 - copy 1024 bytes 
[D] StreamCopy.h : 256 - available: 2048
[D] AudioStreams.h : 459 - GeneratedSoundStream::readBytes: 1024
[D] SoundGenerator.h : 63 - readBytes: 1024
[D] StreamCopy.h : 395 - write: 1024
[D] I2SCodecStream.h : 125 - I2SStream::write: 1024
[D] I2SStream.h : 119 - I2SStream::write: 1024
[D] I2SESP32.h : 98 - size_t audio_tools::I2SDriverESP32::writeBytes(const void*, size_t)
[D] I2SESP32.h : 106 - i2s_write 1024 -> 1024 bytes
[D] StreamCopy.h : 401 - write: 1024 -> 1024
[I] StreamCopy.h : 176 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
[D] StreamCopy.h : 108 - copy 1024 bytes 
[D] StreamCopy.h : 256 - available: 2048
[D] AudioStreams.h : 459 - GeneratedSoundStream::readBytes: 1024
[D] SoundGenerator.h : 63 - readBytes: 1024
[D] StreamCopy.h : 395 - write: 1024
[D] I2SCodecStream.h : 125 - I2SStream::write: 1024
[D] I2SStream.h : 119 - I2SStream::write: 1024
[D] I2SESP32.h : 98 - size_t audio_tools::I2SDriverESP32::writeBytes(const void*, size_t)
````

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-09-03:
Just comment out the assert!
I also [committed a correction ](https://github.com/pschatzmann/arduino-audio-driver/commit/c8e422b590848359243a529a29fa15bbaec08e3f)which replaces the assert with a warning...

pschatzmann, 2025-09-03:
If you don't get any I2C error messages, then the codec setup was working!
I guess you are missing some activation pin: e.g. the PA pin...

pschatzmann, 2025-09-03:
And you forgot the master clock: without it, it will not work...

pschatzmann, 2025-09-03:
[image]

I confirm:
- No other pins are needed.
- GPIO8 is correct

If you define the masterclock, it should hopefully start to work....

dingetje, 2025-09-03:
Victory!

Thanks for your excellent support.

Just for kicks, here's the complete sketch

````
/**
 * @brief We define a custom board w/o any pins and output_device a sine tone
 * with the help of the AudioTools I2SCodecStream
 * @author phil schatzmann
 */

#include "AudioTools.h" // install https://github.com/pschatzmann/arduino-audio-tools
#include "Audioboard.h"
#include "AudioTools/AudioLibs/AudioBoardStream.h"

AudioBoard board{AudioDriverES8311, NoPins};
AudioBoardStream out(board);

AudioInfo info(44100, 2, 16);
SineWaveGenerator<int16_t> sineWave(32000);
GeneratedSoundStream<int16_t> sound(sineWave);
StreamCopy copier(out, sound);

void setup() {
  // Setup logging
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  AudioDriverLogger.begin(Serial,AudioDriverLogLevel::Debug); 

  // initialize i2c because board has no i2c definition
  Wire.begin(15,14); // sda,scl

  DriverPins pins; // no pin definition
  pins.addI2C(PinFunction::CODEC, Wire, true);
  board.setPins(pins);

  // start I2S & codec
  Serial.println("starting I2S...");
  auto config = out.defaultConfig();
  config.copyFrom(info);
  // define your i2s pins if you dont want to use the default pins
  config.pin_bck = 9;
  config.pin_ws = 45;
  config.pin_data = 8;
  config.pin_mck = 42;
  config.rx_tx_mode = TX_MODE;
  config.input_device = ADC_INPUT_NONE; // no input
  config.output_device = DAC_OUTPUT_ALL; // use both channels of the internal dac

  out.begin(config);

  pinMode(46, OUTPUT);    // Configure pin 46 (PA) as output
  digitalWrite(46, HIGH); // Set pin 46 (PA) to HIGH to enable amplifier

  // Setup sine wave
  sineWave.begin(info, N_B4);
}

// Arduino loop - copy sound to out
void loop() { 
  copier.copy(); 
}
````

closed 2025-09-03 by dingetje