pschatzmann/arduino-audio-tools #2175: RTSP Example Compile Error

psimonkey, 2025-09-21:
### Problem Description

I'm trying to compile the communication-generator-rtsp.ino example using platformio.  This is using the code as of commit c589cff earlier today.  I get:

`lib/arduino-audio-tools/src/AudioTools/CoreAudio/AudioTimer/AudioTimerESP32.h:83:12: error: 'esp_timer_restart' was not declared in this scope
       rc = esp_timer_restart(rtsp_timer, time_us);
            ^~~~~~~~~~~~~~~~~`

I've tried changing the code to:

`    esp_err_t rc = ESP_OK;
    if (!started) {
      LOGD("esp_timer_start_periodic");
      rc = esp_timer_start_periodic(rtsp_timer, time_us);
    } else {
      LOGD("esp_timer_restart");
      esp_timer_stop(rtsp_timer);
      rc = esp_timer_start_periodic(rtsp_timer, time_us);
    }
    if (rc != ESP_OK) {
      LOGE("Could not start timer: %d", rc);
      return false;
    }

    initialized = true;
    started = true;
    return true;
  }`

Which compiles, but I don't get any audio output.

### Device Description

ESP32-WROOM-32

### Sketch

```cpp
communication-generator-rtsp.ino from the example.
```

### What is your development environment (incl. core version info)

VSCode, platformio, arduino-esp32 2.0.17.

### I have checked existing issues, discussions and online documentation (incl. the Wiki)

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-09-21:
I am not sure if I want to support any obsolete ESP32 core versions.
I suggest that you use an PlatformIO setup that uses an actual IDF version.
Alternatively you can submit a pull request that uses AudioTimerESP32Legacy.h with obsolete IDF versions

pschatzmann, 2025-09-21:
I committed a correction to use the old logic for IDF <  5.0

psimonkey, 2025-09-21:
Hi Phil - thanks for the quick response.  6331bac fixed my problem.

I was using a fresh install of PlatformIO, which I believe gets you platform-espressif32 6.12.0, Arduino v2.0.17, based on IDF v4.4.7.  It looks like the sitation with Arduino versions and PlatformIO is a little complicated.

It looks like a good solution is using the pioarduino fork, which gets you Arduino 3.3.1 and IDF 5.5.1 with minimal faff: https://github.com/pioarduino/platform-espressif32.

Cheers,